cleanflight/cleanflight #1314: Magnetometer not working

mosi0815, 2015-09-16:
I just tried 1.10.0-RC1 on my spracingf3 today because i want to use softserial ports.
Unluckily the heading just shows 0 in 1.10.0-RC1 After I went back to 1.9.0 everything worked fine again. I tried this a couple of times. Always with the right version of the configurator and clearing all settings.

nwenkel, 2015-09-17:
Same issue here with SPRacing F3 deluxe and Naze 32 (Rev. 5)

HFMan, 2015-09-17:
I just tested this on a standard Naze32, Rev 5.  No sensor data for Mag.

HFMan, 2015-09-17:
Looks to be here:

drivers/compass_hmc5883l.c

    bool hmc5883lRead(int16_t *magData)
    {
        uint8_t buf[6];

        bool ack = i2cRead(MAG_ADDRESS, MAG_DATA_REGISTER, 6, buf);
        if (ack) {
            return false;
    }

The above should be:

**if (!(ack))** {

nwenkel, 2015-09-18:
Confimed! With the mentioned changes working fine again.

xqoperator, 2015-09-18:
the check is introduced in commit c6f5b98a7916a369bcf9445d04feef4036dc3ce5
what other sensors stopped working?

xqoperator, 2015-09-18:
looks like the rest is fine.
I created a pull request with a fix for this

hydra, 2015-09-19:
Fixed in 0d6d83925f6fa5e5f7c2ef1bc5afad1721ef079e

closed 2015-09-19 by hydra