cleanflight/cleanflight #1317: bmp 180 support

handsomejackuk, 2015-09-17:
Just curious is bmp 180 barometer still supported on cleanflight,,,?

SteveAmor, 2015-09-18:
Yes it is. Please can you close this issue (curious question)

stronnag, 2015-09-18:
Yes is is supported (though you need a recent build for external I2C baro). Some of BMP 180 are mis-detected as MS5611 devices. This can be mitigated in the latest builds with the CLI stanza:
````
set baro_hardware = 2
````
Hydra, please close as requested by @handsomejackuk

SteveAmor, 2015-09-18:
Fix for mis-detection in PR #1052

SteveAmor, 2015-09-21:
@handsomejackuk if this information provides and answer to your question, can you please close this issue :)

closed 2015-09-23 by hydra

Tomal2v, 2017-01-16:
BMP 180 not detected on CC3D despite soldering and plugging in the battery correctly 
Help needed