cleanflight/cleanflight #1320: MPU6000 Interrupt Support

henn1001, 2015-09-18:
I´m trying to port the IRQ from the mpu6050.  

https://github.com/henn1001/cleanflight/commit/99e94a818c2379755ed73662b3ff30c08dc178f8

Maybe someone with a little more inside into the codebase can help me out here? It compiles and "runs" fine. It just doesn´t get the interrupt so the watchdog grabs the signal and the loop is at 1100us..
Gyro runs at 1khz. Target is a cc3d as you can see.

Also for later there probably should be one checkGyroDataReady config and not multiples for checkMPU6000DataReady and checkMPU6050DataReady ect..

borisbstyle, 2015-09-18:
I think CC3D needs something like this..

https://github.com/cleanflight/cleanflight/blob/c6f5b98a7916a369bcf9445d04feef4036dc3ce5/src/main/sensors/initialisation.c#L81-L121

henn1001, 2015-09-18:
Yes i already added that part..
https://github.com/henn1001/cleanflight/blob/99e94a818c2379755ed73662b3ff30c08dc178f8/src/main/sensors/initialisation.c#L142

henn1001, 2015-09-18:
Fixed it -  i´ll send the pull to Boris.

closed 2015-09-18 by henn1001