cleanflight/cleanflight #1321: Board - Naze rev6 support

hydra, 2015-09-18:
Add support for this board to Cleanflight.

BMP280 driver already done and waiting to merge.

MPU3050/6000/6050/6500/9150/9250 driver cleanup code is required as the Naze32 rev6 has a 6500 connected via i2c.

Assistance with this welcomed.

borisbstyle, 2015-09-18:
I guess new target needed or you would prefer to have auto detection i2c or spi on mpu6500?

hydra, 2015-09-18:
no new target needed, i am working on the cleanup now.

YFZ450MANNY, 2015-09-18:
Sup fellas! To start off, I wanted to say thank u for all the hard work u guys put in to make our grown-up toys KICK ASS!! lol. The world would be a sad place without u all.. :) And then here comes the stupid question.. More or less when do u think I'll be able to use my rev6 Naze with CF? Judging by the above, I'd dare to say when CF ver 1.11 is released??

hydra, 2015-09-19:
It's booting now.  PR with some changes to follow.

hydra, 2015-09-19:
See #1326 

Binaries will be avilable on the build server shortly.

http://cleanflight.memoryleaks.org/builds/

Nate0624, 2015-09-20:
Hey Gang,

I was able to flash the Rev 6 with the Hex from the nightly updates. However, the Accel doesn't appear to be working. Is it possible I jumped the gun? Can someone point me in the right direction if it is complete and ready for testing?

jcowgar, 2015-09-20:
Flew 5-6 battery packs today on my Naze rev6 w/CleanFlight and all went just fine, no issues found. It even worked right out of the box with my Futaba SBUS, so the Inverter was enabled automatically when I selected SBUS like it should have been.

nolimit01, 2015-09-20:
Jcowgar, where to get this CF for naze32 REV6?

jcowgar, 2015-09-21:
@nolimit01 I do not know. I checked out from git and compiled myself.

hydra, 2015-09-21:
@jcowgar @nolimit01 link in comment https://github.com/cleanflight/cleanflight/issues/1321#issuecomment-141692881

RichDGD, 2015-09-22:
sorry Im a bit computer illiterate. how do i flash the above hex file to the rev6 board? i am using os x and have attempted to compile i got as far as running the arm prompt after adding bin file to my profile in nano but when entering the arm command in terminal it says command not found. is there an easier way, again i apologize if this is a stupid question. I've been driving myself crazy trying to get this board to work with cleanflight. no acc or gyro readings in gui

jcowgar, 2015-09-22:
@RichDGD read the tutorial on compiling in OS X (https://github.com/cleanflight/cleanflight/blob/master/docs/development/Building%20in%20Mac%20OS%20X.md) or look at the links a few comments up. Hydra's comment above yours has a link to the right comment.

nolimit01, 2015-09-22:
It's not letting me save as a hex file.

RichDGD, 2015-09-22:
@nolimit01 I just renamed it as a hex file and now it's working perfectly

nolimit01, 2015-09-22:
Hi RichDGD, I really need some help.

nolimit01, 2015-09-22:
Hey Rich, where to download this file?

Stingersswarm, 2015-09-22:
This might not be the best place to post this but I've been using the white x4r rx with the sbus hack providing an inverted sbus signal. Is there any way to turn off the rev6 serial inverter when using bus for use with the the white x4r? Since the normal sbus isn't available on it.

nolimit01, 2015-09-23:
Got the file installed, re-setup board and the PIDS. Not so good, may have burned up a motor.

Hoppper, 2015-09-27:
Hi!
Have problem enabling feature rx_serial
After rebooting still disabled. Other rx features works fine.

Runfromrobots, 2015-09-29:
I have the same issue as Hoppper ^ - I can enable rx_serial, I can check "serial based receiver" all I want but every time I save or reboot it automatically turns them back off :(

borisbstyle, 2015-09-29:
Maybe you didnt select the proper settings for rx_serial and it resets
because of that

Runfromrobots, 2015-09-29:
@borisbstyle so, first I set serialrx_provider=2 (unless they changed the order of providers I believe this is specifically SBUS), then I enable rx_serial - I've tried putting the commands in a different order, I've tried working in the configuration tab rather than the CLI, and every time I save/reboot it deselects :(

borisbstyle, 2015-09-29:
You need to disable PPM or PWM and than enable serial rx I believe

JasonB91, 2015-09-30:
Having immense problems getting Spektrum Sat (DSM X or DSM 2) to bind with the new Rev6 board. Soldered sat wire direct to the top, from left to right (3.3V, Negative, Signal). 

We're seeing that when using set spektrum_sat_bind = 9, and unplug USB, and power up with 12V input, sat will not initiate bind. Upon reconnect to usb and dump performed on CLI, board defaults back to "0" instead of "9". 

http://www.rcgroups.com/forums/showpost.php?p=32813814&postcount=54

Using this firmware. ^^

At this point, totally lost and beyond frustrated. What am I missing? 

[image]

sjtrny, 2015-10-01:
Any further progress @hydra? I received a whole stack of the rev6 but am waiting for CF support until I ship.

hydra, 2015-10-02:
@JasonB91 no idea without testing, can someone else verify this issue please.

hydra, 2015-10-02:
@sjtrny Still gathering feedback, 1.10.0 had higher priority.   1.10.0 bug fixes come first, then we can look at new features and board support.

JasonB91, 2015-10-02:
@hydra I was able to bind the sat outside the Naze and CF, by binding with an AR6210. It does work with the supported Spektrum port however with the picture I provided.

MJ666, 2015-10-04:
What happens when you reconnect via USB the second time instead of 12V. Since the port is the same as on the REV5 boards I don't see any reason why it should not work with REV6? Not sure if the timing of the bind impulses is affected by the new revision of the board? I don't have one of this yet so I can not test. Have used the binding with many different boards till today and never had issues. I even don't own an full receiver fro alternative binding and using LemonRX DSMX sats in all my copters. I have never tested an original spectrum sat.

JasonB91, 2015-10-06:
@MJ666 We tried both with USB 5V, and direct 3s 12V power. 

http://rotorgeeks.com/index.php?route=product/product&product_id=70

Note at bottom: 

Attention Spektrum satellite users - satellite binding is not currently working with Rev 6 hardware - you must bind your satellite with separately with a receiver before connecting to the Naze board
Instruction on how to bind the satellite directly from the flight control board,   For great value and proven range, check out the Lemon Rx Spektrum compatible satellites.

MJ666, 2015-10-07:
I may need to get one or two Rev6 boards to make it work again. It's working for so many other FC's. It's likely something very trivial.

hydra, 2015-10-08:
driver changes and baro merged to master.

sjtrny, 2015-10-08:
So now all future nightly builds will have experimental rev6 support?

hydra, 2015-10-08:
yes

FAYDID, 2015-10-12:
Hey Hydra, amazing work! Would really like to move from Baseflight to CleanFlight before my first flight on my new quad. Any updates would be awesome. I got most of the way through on the latest version. 

Some issues, could not get gyro, accel or baro to connect and it was pretty unresponsive. Also had some usb connection intermittency. Looking forward to a stable release!

MototechRyan, 2015-10-15:
Thank you for your awesome work Hydra. I'm having an issue when clicking save and reboot. It hangs in reboot and when I click another tab it says "waiting for data". I've waited up to 10 minutes with no change. My only option is to click disconnect. Also note when connecting, the green light blinks the entire time my naze32 rev 6 is connected. I don't mind waiting for a stable release, but am happy to offer any help I can testing binaries.

hydra, 2015-10-19:
@MototechRyan steps to reproduce please.  Sure you flashed it correctly?  full chip erase? video? config dump?

MototechRyan, 2015-10-19:
I think something is wrong with my board. Cause i reflashed with baseflight and its doing the same thing in baseflight. Cancel my issue comment. Ive ordered a rev 5 and will wait for the 1.11 release to use my rev 6 with cleanflight. I will try to get the video and config dump. But feel you development gurus time would be better spent on other issues.

FAYDID, 2015-10-19:
I am having the same issue as @MototechRyan. My board is woking fine however in baseflight. I can post any info needed. Just let me know.

ProDrone, 2015-10-19:
I noticed the same behaviour with my __naze rev5.__
- go to PORTS tab and click on Save and Reboot.
- Log window reports (last displayed lines):
~~~
2015-10-19 @ 23:56:52 -- EEPROM saved
2015-10-19 @ 23:56:52 -- Device - Rebooting
~~~
- Clicking on any other tab after that hangs up waiting for data.
- Disconnect and reconnect gets all back to normal (until steps are repeated - consistently).

Used:
cleanflight master at 526f4777a24e6963ded6e6b92060350aa0e95878
configurator development branch at 7bf70092d35f9e96d1b681c3cc1cda2450049098

borisbstyle, 2015-10-19:
This is a configurator issue right?

I have seen this on all firmwares since the latest configurator.

MototechRyan, 2015-10-19:
The only thing i can think of that changed was i was prompted to update my eprom software....i didnt know the steps to load file locally in cleanflight after right clicking and saving build 2147. So when i tried to open the build by double clicking it in windows, it prompted me to update the eeprom software i have installed for my turnigy 9xr pro tx software. May be irrelevant, but the eprom software was updated by me before figuring out how to load the new build locally in cleanflight.

ProDrone, 2015-10-19:
Don't know. Looks like cleanflight frimware, since i have the same with
configurator master v0.65.0 8e7097c33e64bd4a16b38588caa83fc7c2c14704
and at label v0.65.0 8d39b3f8cc037cf5b7a17b274664c9118238a491

borisbstyle, 2015-10-19:
Even 1.10 official had the same while it wasnt there like 2 weeks ago when
it was released.
I believe @tracernz mentioned something about configurator not closing
connection when disconnecting somewhere.

FAYDID, 2015-10-20:
When I connect with my rev. 6 I am not able to get gyro, accel or baro to connect and it was pretty unresponsive still. I am also not getting any accel movement...

FAYDID, 2015-10-22:
I tired the dev version last night: 2147_805ee8c_travis_cleanflight_NAZE.hex 

The only issue I am seeing in this hex is: In the motor testing only 1 motor and ESC are working. Using BL20a and Cobra motors (in baseflight all work as they should). 

I woud love to switch over to cleanflight. I just need this motor functionality. I also calibrated the ESCs the same way as I did in baseflight. (clicked the motor button to allow them to work, put master to max, plugged in battery, got audio verification, put master to zero, unplugged battery. Plugged battery back in. Only one motor and ESC are working. I tried switching all motors and ESCs as well.)

FAYDID, 2015-10-22:
Okay so very odd. I recalibrated the ESCs again without the battery. Just put the master motor all the way up, music played, put it down, confirmation music played then plugged in the battery and now all motors work.
In baseflight I always connected the battery when the master was all the way up then, all the way down, then disconnected and reconnected the battery and it has always worked.
Odd... but I will get some more props today and try it out today.

MototechRyan, 2015-10-23:
Long story, but i got another new rev 6 in the mail today. Its from a different batch cause my first was purple and this one is red. I connected to baseflight, flashed the most recent stable firmware and its doing the same hanging in reboot when clicking save. So Im pretty sure its not caused by cleanflight cause this new rev 6 has never been connected to or flashed with any cleanflight firmware. Im gonna try reinstalling cleanflight and baseflight on my pc and try a different usb cable and see what happens.

FAYDID, 2015-10-23:
@MototechRyan Ya I decided to just go out and buy and rev. 5 for now and keep the rev. 6 until there is a stable usable version.. The rev. 6 has really pissed me off.

sjtrny, 2015-10-23:
I've been getting reports that it is also affecting rev5 and downgrading Chrome seems to fix the problem.

MototechRyan, 2015-10-23:
Me too, i thought i was ordering a rev 5 and got another rev 6, doh! Ordered a rev 5 for real today. The rev 5's are getting hard to find for a reasonable price online nowadays.

Gonna go see if this new rev 6 has the same missing diode setup that allows 5v to back feed through the 1 thru 6 + pins to the PDB when connected to usb power only

FAYDID, 2015-10-23:
@sjtrny I already tried downgrading chrome didn't fix anything...

FAYDID, 2015-10-23:
@MototechRyan Ya I eneded up just paying a bit more and getting one on Amazon Same Day shipping so I could install today.

andymac4182, 2015-10-24:
@MototechRyan That might be deliberate. I know some of the boards will allow your receiver to be powered up via the USB port so you can configure without requiring the motors to be powered up.

JasonB91, 2015-10-24:
@FAYDID Can you share the Amazon URL? I can't seem to find them myself.

FAYDID, 2015-10-24:
@JasonB91 sure: http://www.amazon.com/gp/product/B014M7XEZO?psc=1&redirect=true&ref_=oh_aui_detailpage_o00_s00

JasonB91, 2015-10-24:
@FAYDID Thanks. I saw that, but assumed it was a "6DOF" Clone of the naze.

FAYDID, 2015-10-24:
I can verify for you. I don't have it yet but should be here soon. How can
I check?

@FAYDID <https://github.com/FAYDID> Thanks. I saw that, but assumed it was
a "6DOF" Clone of the naze.

FAYDID, 2015-10-24:
@JasonB91 It is a "6DOF" Clone of the naze... just FYI...

mattyra, 2015-10-29:
Hey Lads,

Im a little bit wet around the ears on this one, but I'm now running a Naze Rev6 don't care much for the baro... Im have a few issues and was wondering if this is something you are still looking at..

Issue 1 - The angle mode is sloppy and the rates don't affect the speed on pitch and roll (don't care much for this but though i would point this out...

Issue 2 - Cant get black box to work on the built in memory on matter what i do i pick on the Data screen that is there but theres no recording of data present...

Issue 3- time i need more time and good weather to fly

[34 more replies]

closed 2015-12-14 by hydra