cleanflight/cleanflight #1322: SPF3/Kiss30 Calibration issue

Johnno88, 2015-09-18:
I have found  an issue when setting up a clean firmware install, either 1.9 or 1.10. If i go through the normal steps of Firmware, Ports, Config, Receiver, Motors, and then CLI adjustments all seams to work ok.

If however i try to calibrate motors/esc AFTER i have set Looptime=1200,  the motors/esc fail to calibrate.  Further, if i then try to test the motors in this state, i get no reaction until the throttle is approx half, then the motors jump into life at 100% force, in reverse direction and the only way to stop them is to disconnect the main battery.

I have tested  this at different loop time settings, 3500-1600 all seam to work ok, but as soon as i use 1200 in the process, the fault occurs. 

CLI Dumps to follow.

andymac4182, 2015-09-19:
Do you possibly have them enabled for 3d flight?

Johnno88, 2015-09-19:
Thanks for the question, simply no.  I don't think so.

If after the fault is on the quad, i can go into CLI, set the looptime=1600, save and then go back and calibrate the motors/esc and the fault is gone.

andymac4182, 2015-09-27:
Does it do it with OneShot disabled? I am running a looptime of 1000 and it is working with KISS 30amps.

Johnno88, 2015-09-27:
Hi Andrew 

I have moved on to beta flight and now not having any issues.

I don't particularly want to go back and do any more testing of that issue. 

I just thought I should report the issue so the developers would know about it

Johnno

John Whitfield

andymac4182, 2015-09-27:
Simple solution :) I am using betaflight as well. Glad everything worked out for you.

hydra, 2015-10-02:
@borisbstyle specifically, what code might have changed?  I'm not aware of any PR's that fix or change this behavior.

ctzsnooze, 2015-10-02:
Dominic I don't have this exact issue but I and a few others are having issues where after booting the FC via the LiPo, some ESC's just play 'dead' and don't respond at all.  This only happens at relatively low loop times, but is a bit unpredictable and can happen as high as 1000.  Frequently happens around 600 -650.  I don't yet know an completely 'safe' loop time so am now up to 950.  The ESC's can get out of the zombie state only if I power cycle the radio Tx (!), or if I boot the quad with the radio TX off, then switch the radio Tx on. I didn't have this problem until I started trying loop times under 1000.  I'm using a build based on current master as of today.

borisbstyle, 2015-10-02:
@ctzsnooze What you are describing there sounds like a know blheli bug on refresh rates >1khz.
Did you upgrade to blheli 14.1 or 14.2....not sure what the latest was.

@hydra 
Low looptimes can have some very high jitter what can confuse some ESC's and especially ones with auto detections.
I suggest them to try to calibrate with more stable looptimes. Depending on FC and pid controller etc this can be different. Naze32 for example needs >1200
I have done some work in betaflight to make the control loop more efficient and stable so the jitter doesn't occur on lower looptimes. Maybe something for the future.

ctzsnooze, 2015-10-02:
It's with BLHeli 14.1 which is the latest and which is supposed to be fine with loop times under 1000.  What I describe has been documented by several people now.  What seems odd is what I've found, that power cycling the TX restores normal ESC functionality.  

For instance lets say I have a Naze board at loop time 650 and acc_hardware = 1, ESCs nearly always go zombie and unresponsive if LiPo is connected with radio Tx on.  Power cycle the radio, or boot the FC with the radio off, and bingo they are good again.  Nothing else works, but power cycling the radio gets them going every time.  I don't remember this on earlier builds.  Thats why I wonder if it has to do with RC smoothing or the rxfail stuff.  I will test a few things but for now can't pin down where it started.

borisbstyle, 2015-10-02:
@ctzsnooze 
It sounds more like an ESC issue here. You would need to scope it to figure out what's going on.

Johnno88, 2015-10-02:
I have been using betaflight for over a week with no recurring problems until today.  
I loaded this morning hex,
# version
# BetaFlight v2/SPRACINGF3 1.10.0 Oct  2 2015 / 22:36:51 (52172ef)
# dump master

and then tried to calibrate the ESC/Motors with no luck.  

After trying a number of settings, the only way to calibrate was to disable ONESHOT125, and then calibrate.

Previous versions of Betaflight  over the last week have calibrated without this step.

Unfortunately, I am not a tech, but thought it best to provide some feedback incase it is an issue.

borisbstyle, 2015-10-02:
@Johnno88 
In the first screenshot it says Cleanflight 1.10....right?

Johnno88, 2015-10-03:
sorry for your confusion,
I first had the issue when i built a new quad using SPF3 and Kiss30 about two weeks ago. When i finished the build, i had just found out about CF 1.10 and thought to upload the new quad with new firmware. 

I then had the issue and as I am new to quads and thought i should report the issue,

A few days later, i found the beta fight site and thought i would try it to see if the fault followed. You will see from my comment talking to AndyMac that i mentioned i had moved over.

Since moving to Betaflight, I have had no issues until this morning hex update when again the issue appeared.

Not understanding the programming , and because i could get around  the fault by turning off the ONESHOT,  i  thought i should at least put this information forward incase it might mean something more to you.

I am sorry if i am making a mountain out of a mole hill.

Regards
J

Uh1Pilot, 2015-10-18:
I had this same issue, same board, same clean flight version. Calibration at a loop time of 1000 put the KISS 18A esc's in 3D mode.

Uh1Pilot, 2015-12-15:
Just an update, had to replace a motor so I decided to upgrade them all, just tried to re calibrate and the 3d mode issue is still happening with KISS 18A esc's; looptime @1000, and Oneshot125 enabled. It's kinda a safety issue because if you're not ready for negative trust after arming it, the quad can fuck some shit up.

closed 2017-02-25 by hydra