cleanflight/cleanflight #1323: CC3D wrong mapping of outputs in mixer AIRPLANE modes

sppnk, 2015-09-19:
Hi, with latest release 1.10 RC, I have found that the outputs for the servos are wrong with mixer AIRPLANE. With yaw stick I move elevator, and with roll stick I can move rudder. Motor is OK.

The only way to correct this is by CLI:

mixer customplane
smix reset
smix 0 2 0 100 0 0 100 0
smix 1 3 0 100 0 0 100 0
smix 2 4 0 100 0 0 100 0
smix 3 5 0 100 0 0 100 0

servo 3 1000 2000 1500 90 90 -100 -1
servo 5 1000 2000 1500 90 90 -100 -1 
(these last commands to reverse my servos)

In 1.8.1 it worked well, so something have been changed in the code, but I havent been able to find the problem.

hydra, 2015-09-23:
Please can you show us how you've got your servos connected.  A photo would be good.

hydra, 2015-09-23:
I will look at this, since this issue appears to be a give a good overview of the problem.  I will close the other issues relating to the problem.  Please keep comments focussed.

sppnk, 2015-09-23:
Thx, the servos are conected like this (same in 1.8.1, wich is working):

|CC3D OUTPUT|SERVO/MOTOR|
|--------------------|----------------------|
|1|ESC(MOTOR)|
|2|not connected|
|3|AILERON R|
|4|AILERON L|
|5|RUDDER|
|6|ELEVATOR|

eijowk, 2015-09-27:
I have exactly same behaviour as sppnk in my airplane with naze32. I was running 1.9.0 before with correct behaviour.

sppnk, 2015-10-06:
Changing title because of eijowk comment. Thx.

Basti77, 2015-10-15:
The current situation with CF and CF 1.10.0 Config Creator V0.66
Default channel mapping
MotorsTab:
      servos
      1 Throttle
      2 No function
      3 No function
      4 ROLL
      5 ROLL
      6 YAW
      7 PITCH (But there is no connection for servo 7 on the CC3D)
Once you changed something in the Servo tab, only the servos at output 4 works.
Then only help is to Restore default settings.
How can I put Pitch on output 2 or 3?

SlyMike, 2015-10-15:
Same problem here, if I set my servos as servo 2,3,4,5 on pins 3,4,5,6 (with one motor defined, CUSTOMAIRPLANE).... I end up controlling servos 3,4,5 and 6 in the GUI.

# smix
Rule	Servo	Source	Rate	Speed	Min	Max	Box
#0:	2	0	100	0	0	100	0
#1:	3	0	100	0	0	100	0
#2:	4	2	100	0	0	100	0
#3:	5	1	100	0	0	100	0

My hope was to be able to configure the following:

Rule	Servo	Source	Rate	Speed	Min	Max	Box
#0:	2	0	100	0	0	100	0
#1:	3	0	100	0	0	100	0
#2:	4	2	100	0	0	100	0
#3:	6	1	100	0	0	100	0

pin    function               designation
1	MOTOR               servo 1 (ESC)
2	AILERON             servo 2
3	AILERON             servo 3
4	RUDDER             servo 4
5	(LED_STRIP)      LED_STRIP
6	ELEVATOR         servo 6

.... However servo 2 will not move in this configuration. ESC and LED_STRIP work fine.

(v1.10, 0.66, CC3D ATOM)

Basti77, 2015-10-16:
OK I understand it slowly.
Here my intermediate result:
The setup is for a twin-engined aircraft.
First you have to assign the forces the outputs over CLI.
Sorce 0 = Roll (force)
Sorce 1 = pitch Force)
Sorce 2 = Yaw (force)
Servo = Physical Slot -1

smix 0 2 1 100 0 0 100 0	# elevator
smix 1 3 0 100 0 0 100 0	# aileron
smix 2 4 0 100 0 0 100 0	# aileron
smix 3 5 2 100 0 0 100 0	# rudder
smix 4 0 2 100 0 0 100 0	# Motor 1
smix 5 1 2 100 0 0 100 0	# Motor 2

servo 1 1000 2000 1500 90 90 -100 -1 # turn the Yaw Force outpot on Motor 2

mmix 0 1 0 0 0 # enabe Thr Output
mmix 1 2 0 0 0 #enable Thr Output
theoretical works and on the desk

save!
Then assign the outputs via the servo tab control functions.
[image]

eijowk, 2015-10-19:
I don't get this... how should I set up the mixers to work on a typical airplane with one motor, ailerons, rudder and elevator?

And if I want to add flaperons from an Aux channel how do I set up that?

digitalentity, 2015-10-22:
With the above referenced PR mapping is again as it was in 1.8.1

CC3D OUTPUT | SERVO/MOTOR
------------ | -------------
1 | ESC(MOTOR1)
2 | ESC(MOTOR2)
3 | AILERON R
4 | AILERON L
5 | RUDDER
6 | ELEVATOR

SlyMike, 2015-10-22:
I would still like to see CC3D Output Pin 2 mappable as either ESC OR SERVO to allow the use of 1xESC,  4xSERVO and Function LED_STRIP enabled in CUSTOMAIRPLANE configuration.

I have realised my previous mistake of configuring my setup as servos 1-6 when it should have been 0-5.

It seems that CC3D Output Pin 2 (servo1) is ESC or nothing at the moment.

sppnk, 2015-10-23:
Thx, I'll try ASAP !!

creol, 2015-12-30:
I dont want to hijack this, but I am having a similar issue with CC3D Quad setup. The Yaw and Pitch are mixed up. Could this be same issue?

Zero9r, 2015-12-31:
Creol that can be changed on your radio or unter the radio tab.  where it says your mapping.  for example  "TAER1234"  Just switch the "R" and "E"  so "TARE1234"

kressir, 2016-01-19:
I have the same issue running a Flip32 (NAZE).  I had 1.09 running fine and updated to 1.11.  I did a backup on 1.09 and restored in 1.11.  After the update, the servos were not responding correctly - they were moving the wrong control surfaces.  Throttle was OK.
 
[image]

I reset the configuration and selected the Airplane mixer (as it was before the reset). Results were the same - wrong servos move.  The throttle still worked OK.  It seems like the servo mixing is broken, but the motor mixing is working in the Airplane standard mixer.
I then selected the Custom Airplane mixer and used the example smix settings in the documentation (https://github.com/cleanflight/cleanflight/blob/master/docs/Mixer.md - at the bottom of the page)...
 
[image]

```
mixer CUSTOMAIRPLANE
mmix reset
mmix 0 1.0 0.0 0.0 0.0  # Main Engine  <-- I edited this for my setup, single engine

smix reset
# Rule  Servo   Source  Rate    Speed   Min Max Box
smix 0 3 0 100 0 0 100 0  # Roll / Aileron
smix 1 4 0 100 0 0 100 0  # Roll / Aileron
smix 3 5 2 100 0 0 100 0  # Yaw / Rudder
smix 2 6 1 100 0 0 100 0  # Pitch / Elevator  <-- I changed source from 2 to 6 for my setup
```

These settings had the same effect as the Airplane mixer.  I did need to change the last line to a 6 instead of 2 for a source.  I have my plane set up with elevator on output 6.  
 
[image]

I set it up according to the motor order image displayed in the configurator when the Airplane mixer is selected.  This worked with the 1.09 Airplane mixer.

I was playing in the servo tab to identify the servos by number and I noticed that the servos seem to be zero based.  This numbering scheme does not match the motor order image as that seems one based.
 
[image]

On a hunch, I ran the same smix commands as above, just decremented the servo number by one in all four commands like this:

```
mixer CUSTOMAIRPLANE
mmix reset
mmix 0 1.0 0.0 0.0 0.0  # Main Engine
smix reset
smix 0 2 0 100 0 0 100 0  # Roll / Aileron
smix 1 3 0 100 0 0 100 0  # Roll / Aileron
smix 3 4 2 100 0 0 100 0  # Yaw / Rudder
smix 2 5 1 100 0 0 100 0  # Pitch / Elevator  
```
It worked.  It seems that there is a zero base to one base error somewhere.  Possibly limited to Airplane, Custom Airplane, and Flying Wing mixes.

The Configurator has some of this going on.  There is some ambiguity between the motor order image and the servos tab.  The Motor tab also displays the servos in one base format.
 
[image]

I now have the smix command figured out and prefer it to the Airplane standard mix, but I still think we should get the code fixed so that an update doesn’t require re-wiring to keep the standard Airplane mixer.

One advantage to using the smix command is that I can use flapperons.  My final setup looks like this:
```
# mixer
mixer CUSTOMAIRPLANE
mmix reset
mmix 0 1.000 0.000 0.000 0.000
smix reset
smix 0 2 0 100 0 0 100 0  #Left Aileron
smix 1 3 0 100 0 0 100 0  #Right Aileron
smix 2 4 2 100 0 0 100 0  #Rudder
smix 3 5 1 100 0 0 100 0  #Elevator
smix 4 2 9 -30 0 0 100 0  #Left Flapperon
smix 5 3 9 30 0 0 100 0  #Right Flapperon
```
Rules 4 and 5 give flapperon functionality to the servos 2 and 3 using AUX 2 at 30%.

closed 2017-02-25 by hydra