cleanflight/cleanflight #1324: CC3D sensor stops readings after configure and reset

sppnk, 2015-09-19:
I have been trying to configure airplane with 1.10 RC  and other experimental builds based in 1.10 master branch. 

After other problems almost solved (#1323), I am able to configure my servos and test everything works ok in my bench. The servos respond to my transmitter sticks, and I can arm by switch and test that ANGLE mode works ok (the control surfaces move when plane does).

But If I switch off everything, disconnect Lipo and conect again, the servos respond to transmitter, but ANGLE or HORIZON modes don't work anymore.

I go then to the configurator or EZGUI by bluettoth port and can see that the accelerometer and gyro sensor graphic is showing stable/static values of the sensors. It seems the i2C readings stop.

I have reproduced the issue several times, with 1.10RC and others like digitalentity branch for fixed wing.

hydra, 2015-09-23:
@sppnk please stay focussed and report only one problem per ticket.

There is specific retry code for sensor detection for CC3D.  Some boards respond quicker than others.  The CC3D does not use I2C for sensors.

There are known issues with EZGUI with regards to restarting and bluetooth connections; last time i looked it was problematic.

Feel free to experiment with the CC3D startup code, perhaps adding some extra delays before the sensor detection between retries, if you find something that works please create a PR.

hydra, 2015-09-23:
I will close this, likely the problem is with the EZ-GUI bluetooth connection.  If not please open a new ticket and keep it focussed with exact steps to repeat.

sppnk, 2015-09-24:
OK, because the main issue was that the ANGLE mode stops working,  I close this and open more fosussed issue #1337

closed 2015-09-24 by sppnk