cleanflight/cleanflight #1327: CF 1.10 RC2  Serial RX Failsafe issue

Tabsy123, 2015-09-19:
If you want the motors to all stop but still have a few seconds chance at regaining control it seems you can not do this like you could on 1.8.1

When radio is turned off ( out of range)  and failsafe is set to 1000  (esc calibrated low point on my build is 1050)   The motors all stop for about a second then the craft will bring on the front or back etc as if its trying to level it self even tho the other 2 motors are off (as they are asked to at a value of 1000)  
Would be good if all motors turned off when failsafe is set to a value bellow your esc low point value   
Eg:  1000 failsafe    ,  1050 low point esc ,  motors should all stop and stay stopped unless radio signal has come back before the failsafe time out etc.

Tabsy123, 2015-09-20:
All good I figured out how to get what i wanted,  i just disable failsafe ( untick it)  then it behaves exactly as i want  :)

hydra, 2015-10-02:
great, closing, thanks for feedback!

closed 2015-10-02 by hydra

ducktaperules, 2015-10-02:
This issue sounds like some weirdness i experienced with the fail-safe.
IIRC when fail safe is triggered the armed but low throttle period that happens before the copter disarms does not seem to respect min_throttle, causing motors to go below min_throttle and stall out.