cleanflight/cleanflight #1329: CC3D Hardware - Output Channel mapping

Basti77, 2015-09-20:
The mapping of the output channels is wrong.
There is no elevator in the "Plane" Preset.

Output 2 and 3 have no Servos function.  
Output 4 and 5 are the aileron servos
Output 6 is Ruder,
But where should the elevator servo connected?

The control function of the transmitter "Pitch" is correctly recognized as a pitch.

sppnk, 2015-09-20:
Hi Basti, check issue #1323. It is an old problem unfixed yet.

If you are able to setup CC3D throught CLI commands like me (explained in #1323) you won't be able to fly because of issue #1324.

I hope this could be solved soon, in order to test https://github.com/digitalentity/cleanflight version with future GPS navigation support por airplanes (beta-testers needed there !!).

hydra, 2015-09-23:
Duplicate of #1323 - please keep conversation focussed, search before opening new issues.

closed 2015-09-23 by hydra