cleanflight/cleanflight #1330: Current Monitor Input Selection of PA5 (ADC1_IN5) / Bottom ADC Pad "E"

dhenrie0208, 2015-09-20:
I'm using the Naze32 Rev5 board with Cleanflight firmware "1.9.0 NAZE 2015-6-2 0:41 (stable)", and I've built a basic current monitor that consists of a 2W 0.001 Ohm resistor (=44.7A max) between a 3S battery positive output and the load (the drone). The terminals of the resistor are connected to a simple op-amp circuit that removes the common mode voltage, so that the op-amp output voltage is equal to the voltage across the resistor. This setup produces 1mV per amp the drone draws with 0V DC bias - 0A = 0mV, 10A = 10mV, 20A = 20mV, etc... The schematic and simulation is viewable [here](http://falstad.com/circuit/circuitjs.html?cct=$+1+0.000005+10.200277308269968+50+5+50%0AR+256+432+256+368+0+0+40+12+0+0+0.5%0Ar+256+432+256+496+0+0.001%0Ai+256+496+256+576+0+33%0Ag+256+576+256+608+0%0Ar+256+432+368+432+0+100000%0Ar+256+496+368+496+0+100000%0Ar+368+416+480+416+0+100000%0Ar+368+512+480+512+0+100000%0Aa+368+464+464+464+1+12+0+1000000%0Ag+480+416+512+416+0%0Aw+368+480+368+496+0%0Aw+368+448+368+432+0%0Aw+368+432+368+416+0%0Aw+368+496+368+512+0%0Aw+480+464+480+512+0%0Aw+464+464+480+464+0%0AO+480+464+576+464+1%0AO+256+432+208+432+1%0AO+256+496+208+496+1%0AO+368+416+368+368+1%0AO+368+512+368+560+1%0Ax+611+472+688+475+0+24+to+ADC%0Ax+115+468+242+471+0+12+Current+Sense+Resistor%0Ax+233+347+269+350+0+24+Bat%0Ax+231+648+275+651+0+24+-Bat%0Ax+363+631+655+634+0+12+op-amp+(LM2902N,+single+supply)+powered+by+battery%0Ax+158+538+235+541+0+12+ESCs+/+Motors%0Ao+1+64+0+34+0.03653754093327257+23.945242826029514+0+-1%0Ao+16+64+0+34+0.03653754093327257+9.765625e-55+1+-1%0A).

Because I have a PWM-only receiver (Turnigy 9x8Cv2) and am using GPS, I want to avoid sacrificing another channel to connect the current monitor output to Pin 10 / ID 8 of the receiver port as indicated in the [Cleanflight 'Board - Naze32' documentation](https://github.com/cleanflight/cleanflight/blob/27db9ad0fe9a4d8abc05c59db201b8944324eea4/docs/Board%20-%20Naze32.md).

The Naze32 Rev5 has an ADC pad broken out on the bottom (labeled "E" on page 4 of the [Abuse Mark manual](http://www.abusemark.com/downloads/naze32_rev3.pdf)). In [src/main/drivers/adc_stm32f10x.c](https://github.com/cleanflight/cleanflight/blob/27db9ad0fe9a4d8abc05c59db201b8944324eea4/src/main/drivers/adc_stm32f10x.c), it is acknowledged that this Rev5 ADC breakout "Pad 'E'" exists, and support code for this pad is found in the files shown below. __I'd like to connect the output of my current sensor to this pad so I can keep Pin 10 / ID 8 of the RC port as a PWM channel input.__

The code that assign the ADC channel input to the current monitoring code is given in:

[src/main/target/NAZE/target.h](https://github.com/cleanflight/cleanflight/blob/1f127905f70f0388449d212892d546cb50f4a8cc/src/main/target/NAZE/target.h):

```c++
#define CURRENT_METER_ADC_GPIO      GPIOB
#define CURRENT_METER_ADC_GPIO_PIN  GPIO_Pin_1
#define CURRENT_METER_ADC_CHANNEL   ADC_Channel_9
```

[src/main/drivers/adc_stm32f10x.c](https://github.com/cleanflight/cleanflight/blob/27db9ad0fe9a4d8abc05c59db201b8944324eea4/src/main/drivers/adc_stm32f10x.c):

```c++
#ifdef CURRENT_METER_ADC_GPIO
    if (init->enableCurrentMeter) {
        GPIO_InitStructure.GPIO_Pin   = CURRENT_METER_ADC_GPIO_PIN;
        GPIO_Init(CURRENT_METER_ADC_GPIO, &GPIO_InitStructure);
        adcConfig[ADC_CURRENT].adcChannel = CURRENT_METER_ADC_CHANNEL;
        adcConfig[ADC_CURRENT].dmaIndex = configuredAdcChannels++;
        adcConfig[ADC_CURRENT].enabled = true;
        adcConfig[ADC_CURRENT].sampleTime = ADC_SampleTime_239Cycles5;
    }
#endif
```

All of the above code is replicated for "Pad 'E'" for ADC12_IN5 as well (`enum ADC_EXTERNAL1`).

The code where the ADC values are used for monitoring current is the following:

[src/main/drivers/adc.h](https://github.com/cleanflight/cleanflight/blob/27db9ad0fe9a4d8abc05c59db201b8944324eea4/src/main/drivers/adc.h)

```c++
typedef enum {
    ADC_BATTERY = 0,
    ADC_RSSI = 1,
    ADC_EXTERNAL1 = 2,
    ADC_CURRENT = 3,
    ADC_CHANNEL_MAX = ADC_CURRENT
} AdcChannel;
```

[src/main/sensors/battery.c](https://github.com/cleanflight/cleanflight/blob/27db9ad0fe9a4d8abc05c59db201b8944324eea4/src/main/sensors/battery.c)

```c++
    switch(batteryConfig->currentMeterType) {
        case CURRENT_SENSOR_ADC:
            amperageRaw -= amperageRaw / 8;
            amperageRaw += (amperageLatestADC = adcGetChannel(ADC_CURRENT));
            amperage = currentSensorToCentiamps(amperageRaw / 8);
            break;
```

__Can the option to select between "RC Pin 10 / ID 8" and "Pad 'E'" be made available either through the Cleanflight GUI or CLI?__ I envision a change similar to the following:

[src/main/sensors/battery.c](https://github.com/cleanflight/cleanflight/blob/27db9ad0fe9a4d8abc05c59db201b8944324eea4/src/main/sensors/battery.c)

```c++
    switch(batteryConfig->currentMeterType) {
        case CURRENT_SENSOR_ADC:
            amperageRaw -= amperageRaw / 8;
            if(currentSensorUseExtADCPad) {
                amperageLatestADC = adcGetChannel(ADC_EXTERNAL1);
            } else {
                amperageLatestADC = adcGetChannel(ADC_CURRENT);
            }
            amperageRaw += amperageLatestADC;
            amperage = currentSensorToCentiamps(amperageRaw / 8);
            break;
```

where `currentSensorUseExtADCPad` would be a boolean controllable via the CLI / GUI, like so:

`set currentSensorUseExtADCPad=1` - use External ADC Pad "E" as current sensor input
`set currentSensorUseExtADCPad=0` - use RC Port Pin 10 / ID 8 as current sensor input (default)

Thank you for your great work and consideration. I apologize for 'yelling' with the bold font, but I wanted to easily identify the purpose of this message.

ledvinap, 2015-09-20:
Your level-shift circuit won't work. Try changing one of 100k resistors by 0.1% - resulting error will be huge. Temperature effects, OpAmp input bias and input offset will have similar results ... 

As for the code .. it is probably easier to change target.h and recompile if you are the only one using it. 

Supporting user-selectable ADC pin mapping is IMO possible (I am working in it, but it is not my priority) and would be cleaner ..

Pierre-A, 2015-09-20:
Keep in mind that the ADC is broken in current version due to an incorrect DMA index for RSSI and External (sse PR #1301).

Pierre-A, 2015-10-12:
PR #1301 has been merged so DMA index is now correct for RSSI and External.

closed 2017-02-25 by hydra