cleanflight/cleanflight #1331: Decouple profile reference from dumps.

readerror67, 2015-09-21:
With the current implementation, its required you change profiles before dumping. Then the dump includes stuff to change profiles in the output, the problem is if you would like to duplicate a profile this involves copying everything in to a text editor, finding random lines which contain the profile number and changing before re-pasting in to another profile.

Since its required to manually change profiles before dumping and restoring, it would make much more sense to not reference the profile within the dump. This would make it easier for folks to copy one profile to another easily.

SteveAmor, 2015-09-21:
I believe this is a clean flight configurator issue and should probably be raised there?

readerror67, 2015-09-21:
Hm no.

The CLI is part of the FC software.

stronnag, 2015-09-21:
I really don't see the problem here, in my opinion, having the single profile reference in the dump is beneficial, as it allows the user to reference the encapsulated settings to the profile. I see no good reason to remove it.

readerror67, 2015-09-21:
The reason to remove it is because its not intuitive for users, it requires manually sorting through 348 lines of text to change 1 line (which in mine is line 210). Wouldn't it make more sense to tag the profile in a comment instead?

If you want to duplicate profile 0 to profile 1 to test something you can see how this is a cumbersome process. By removing it, its doable with an Android phone and the copy/paste feature.

stronnag, 2015-09-21:
However it is useful for other use cases, such as automated backup and restore.

readerror67, 2015-09-21:
Yea so you can easily append that within your 3rd party tool then. I'm talking about usability for the 99%.

hydra, 2015-09-23:
tools should not be written to parse or otherwise inspect cli output.  So no @readerror67, @stronnag should NOT do that.

closed 2017-02-25 by hydra