cleanflight/cleanflight #1332: AUX settings not restored with pasted dump.

readerror67, 2015-09-21:
In reference to #1331 
Apparently when the profile is changed just before the aux settings are pasted, it causes a delay in switching profiles (even if you are on profile 0 and pasting a dump that contains 'profile 0').

@hydra mentioned this is due to the lack of flow control which makes sense.

SteveAmor, 2015-09-21:
I believe this is a clean flight configurator issue and should probably be raised there?

stronnag, 2015-09-21:
In my experience aux profiles are restored, even when the profile is changed, *as long as* sufficient care is taken to ensure items are pasted in small batches, with delays to allow the poor old FC to keep up.

Arakon, 2015-09-22:
Whenever you paste anything containing a profile switch in it (i.e. "profile 2"), it takes a brief moment as the FC switches over. Anything pasted in that brief moment gets lost. Best way is to paste up to a "profile x" command, then continue pasting the next section afterwards to give it time to perform the switch..

readerror67, 2015-09-22:
Or the useless profile reference within the dump can be removed per #1331 and things will 'just work(tm)'

borisbstyle, 2015-09-22:
just paste AUX section once more :D

hydra, 2015-09-23:
Duplicate of #325, opened about 1000 issues ago.

closed 2015-09-23 by hydra