cleanflight/cleanflight #1333: MSP_RX Issue

khockuba, 2015-09-22:
When using MSP_RX as reciever in Cleanflight the reported values are completly gibberish and some of them stucked on 855, quad is basicaly non controllable, on 1.9.0 it was perfectly fine.

hydra, 2015-09-23:
Oh wow!  Someone is using MSR_RX! :smile:  @ProDrone This probably needs your attention.

ProDrone, 2015-09-23:
Oh, well, what is it used for? I see in http://www.multiwii.com/wiki/index.php?title=Multiwii_Serial_Protocol
a reference to "UART radio projects", which seems to be here: http://www.multiwii.com/forum/viewtopic.php?f=8&t=3101&start=100 and appears to be an abandoned project...

Maybe we keep it for backward compatibility then? Correct me if i'm wrong but my solution would be to remove it from the code.

conkerkh, 2015-09-23:
I've programmed my own UART reciever based on esp8266. Apart from the fact I'm using it, I would say that this solution is really coninient since you can also get telemetry out of this thus leaving all other input pins for something else. If you don't wanna keep it please point me to the code that is used there. I'll compile my own firmware

atomiclama, 2015-09-23:
@conkerkh that is exactly what I'm in the middle of developing. Correct me if I'm wrong but can't you just use the MSP command to set the RCcommand or have I missed something there.

Do you have any details / web links to what you are doing. See if I have missed something, get some ideas. Cheers

conkerkh, 2015-09-23:
Well yes but still it has to be selected in configuration tab in clean flight at least that's what I found to be working. Actually only sending stuff is fairly easy job. My work to be honest is based on loads of other infos found on the Internet. I can send you simple C function used to update rc values. I might release the code on GitHub but it needs more work because it's Woking efficiently however it's pretty horrible to look at and work on:P

atomiclama, 2015-09-23:
Thanks but I have plenty of code to use ;-)
Most from internet some of my own.
I'll check this over if no one else fixes it, once I get to this stage. 
Cheers

conkerkh, 2015-09-23:
Well on 1.9 it was ok so you can just roll back till someone fixes this

ProDrone, 2015-09-23:
I see there is still interest in this feature, so i will look into this problem. Is there a simple app that provides stick and switch input using msp commands?

atomiclama, 2015-09-23:
Not that I have found yet, sorry

conkerkh, 2015-09-23:
Unfortunately no at the moment I was trying to do this however I found it a bit unreliable due to lack of feedback what you are actually pressing. Impossible to fly this way. Maybe only for switching modes

hydra, 2015-09-26:
Fix merged.

PR #1343 (2015-09-26): Fix RX_MSP oscillation, > 8 channel support
The implementation of `MSP_SET_RAW_RC` wrote the read rc values directly into `rcData[]`. In rx.c's `readRxChannelsApplyRanges()`, these values were read from `rcData[]`, then channel mapping was applied to juggle the channels around, then the values were stored into `rcRaw[]`. In `detectAndApplySignalLossBehaviour()`, if these values were considered reasonable then they were copied back to `rcData[]`.

The issue is that if the channel mapping was not set to the identity map for MSP ("AERT1234"), and `readRxChannelsApplyRanges()` was called again bef [... cut]
closed 2015-09-26 by hydra