cleanflight/cleanflight #1335: CF 1.10 RC1 - flight glitches

alvins82, 2015-09-22:
I'm not sure this is the right forum - please close if not.

Running through some basic flight tests after flashing and experienced some major glitches (almost resulting in crashes). On a couple of the tests one of the motors just throttled up for no reason resulting in the quad lurching to one side - I had to be super quick to cut throttle. Happened twice and I stopped teh tests.

Blackbox attached. Flashed back to 1.9 - no issues there.

https://www.dropbox.com/s/fd8fvqxnh4cykbo/blackbox_log_2015-09-19_161837.TXT.zip?dl=0

borisbstyle, 2015-09-22:
Where in the log was the glitch? Part with no throttle input?

alvins82, 2015-09-23:
Sorry should have included - edited above.

hydra, 2015-09-23:
Using FRSky SBUS RX with old FrSky firmware?

alvins82, 2015-09-23:
Ezuhf with latest stable.

hydra, 2015-09-23:
@alvins82 thanks for RX details. @borisbstyle Not had the chance to look at the logs, did you spot anything?

borisbstyle, 2015-09-23:
@hydra 
Nothing weird what could come from the software. I do see sudden drop of throttle like there was a failsafe. I would say it is coming from the rx if its the failsafe.

borisbstyle, 2015-09-23:
@alvins82 
Can you produce more logs perhaps? Confirm that its an issue.

alvinpb, 2015-09-23:
@borisbstyle  which log did you see sudden drop of throttle?

The two areas that I had issues with - my throttle input was the same before a quick throttle down (which is me saving the quad). From my uninformed eyes - it looks like the gyro somehow gets 'scrambled' causing the motors to react in response.

@hydra  - Will do on Saturday with RC2

[image]
[image]

borisbstyle, 2015-09-23:
@alvinpb 
O yeah I see a glitch there. It looks like it hangs there. There is a 9ms gap between loop iteration 1626 and 1627.
This means your copter misses 9ms of controlling and has to catch up so you get this burst.

Can you post more detaills about your config?

borisbstyle, 2015-09-23:
That was in log 5/5

alvinpb, 2015-09-23:
@borisbstyle this my dump on 1.9 https://dl.dropboxusercontent.com/u/7607766/old.txt.

- Naze32 acro (added baro)
- m8n gps + mag
- Afro30 esc
- SunnySky 2216 1100KV on QAV400
- Ezuhf Diversity RX
- ImmersionRC 5.8ghz vTX

Let me know what other info I can provide.

borisbstyle, 2015-09-23:
@hydra 
What version is the RC1 actually? From what commit is it compiled?

@alvinpb 
Can you type version in cli to shiw git repo number?

alvinpb, 2015-09-23:
@borisbstyle I did take a dump on the RC1 (I flashed back to 1.9 after my issues above).

https://dl.dropboxusercontent.com/u/7607766/new.txt

borisbstyle, 2015-09-23:
@alvins82 
Why did you decide to increase the looptime on new version to 3500? It was 2500 before.

alvinpb, 2015-09-23:
@borisbstyle I don't think I did - that was straight after flash as far as I can remember.

alvinpb, 2015-09-23:
@borisbstyle I think I misunderstood your comment. That dump was straight after a flash. I can't remember if for the test I was running 3500 or 2500.

borisbstyle, 2015-09-23:
@alvinpb 
Yes you are right. It was indeed 2500 in the logs. Log 5/5 is very clear, but th elog 2/5 just shows a jump of 100us and we see the glitch. 

I will try that version with your settings over here on naze rev5.

In the meanwhile. What kind of baro did you add it btw? Could you test with baro disabled as well
set baro_hardware = 1

alvinpb, 2015-09-23:
@borisbstyle I soldered an MS5611 (same as Naze32 full). Will have to run tests on Saturday.

borisbstyle, 2015-09-23:
@alvinpb 
No problems found here. I did use the latest RC2, but not much different I guess.

borisbstyle, 2015-09-28:
@alvinpb 
And any tests?

hydra, 2015-09-29:
i suspect i2c bus timeout, likely caused by the addition of the user-soldered MS5611.

hydra, 2015-09-29:
https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/bus_i2c_stm32f10x.c#L85

might be worth logging i2c timeouts via blackbox

@thenickdude should be able to help there.

https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/bus_i2c_stm32f10x.c#L101

https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/bus_i2c_stm32f30x.c#L74-L78

hydra, 2015-09-29:
I don't think this particulat issue is a fault in the RC, removing milestone.

borisbstyle, 2015-09-29:
@hydra 
I noticed some huge cycletime jumps during baro reads anyway in the code. I also think the sequence of things in the code should be redesigned anyway to minimize or eliminate jitter.

alvinpb, 2015-09-29:
hey guys - sorry I haven't had a chance to test it - I will make time tomorrow night. I think you guys are on the right track as I when I ran the tests previously in acro mode I didn't experience the issue. Will update when I have some test results.

alvins82, 2015-09-30:
I ran RC3 tonight and I didn't experience any issues. One thing I did different was flash with the dev configurator and that I suspect was a mistake on my part when I tried RC1 (I'm not sure if that explains the glitching seen the blacbox). I will run some more tests over the next couple of days and report back.

So far so good.

hydra, 2015-10-02:
ok, closing this issue.  not much we can do right now.

@borisbstyle is right, there are things we can do to improve the scheduling of work that the FC needs to do.  We can work out a plan and raise appropriate tasks - we will reference this when we do.

closed 2015-10-02 by hydra