cleanflight/cleanflight #1337: CC3D sensors stop sending values

sppnk, 2015-09-23:
@Hydra, In version 1.10  I can configure airplane throught CLI, and ANGLE or HORIZON modes work fine, but If I switch off power,and conect again, the servos respond to transmitter, but ANGLE or HORIZON modes don't work anymore.

hydra, 2015-09-26:
Please provide configuration dump.  A video here might be nice to.  I don't see what CLI configuration has to do with anything?  and when you say 'works' what do you mean?

This is a terrible bug report, I have so many questions I don't even know where to begin.   For this to be investigated we need you to provide exact steps to repeat after resetting your FC to defaults using latest release candidate (rc2).

hydra, 2015-10-02:
I'm closing this, no response.

closed 2015-10-02 by hydra