cleanflight/cleanflight #1345: PID's change inflight with 1.10 RC

borisbstyle, 2015-09-25:
Users who upgrade from 1.9 reported change of PID's in flight after upgrading to 1.10. 
This has to do with changed AUX failsafe behaviour in combination with inflight adjustments

See here for fix:
https://github.com/cleanflight/cleanflight/pull/1339

hydra, 2015-09-26:
Likely inflight adjustments should not be applied if rxloss is detected.  That seems like the proper solution to me unless I'm missing something.

hydra, 2015-09-26:
Hmm, there is already a check.  This needs a bit more investigation.

@ProDrone can you look into it?

borisbstyle, 2015-09-26:
Isn't it just easier to set aux failsafe to 1500 to have old behaviour like I did in PR #1339
Users have the ability to change anyway.

borisbstyle, 2015-09-26:
I would suggest to go 1500 fixed asap and release new rc till the automatic behaviour is fixed.
Too low PIDs lead in crashing and uncontrolled vehicle behaviour and there was already an issue openend what could be related to this.

I actually have many reports from people with this behaviour.

ProDrone, 2015-09-26:
@hydra It has my attention. I already found several issues in the initialization. I am not in favor of changing the original intent of the code, it needs to be solved where the problems occur. Also giving a mid-position to a two position stick is not logical. On top of that is any position that follows a previous-other position a switching event. There are some more (related) issues, so it needs more investigation...

hydra, 2015-09-27:
fix merged, @borisbstyle please examine @ProDrone's fix, if you could also verify the latest master works as you expect now that would be great.

Open a new issue if there's still a problem and reference this one as required.

@ProDrone thanks for chat on IRC - really helps.

closed 2015-09-27 by hydra

borisbstyle, 2015-09-28:
Ok I see that his approach is better as it keeps the initial intention of this feature.

PR #1351 (2015-09-27): RX init issue, optimization.
This should fix #1345, keeping the original / intended functionality as it currently is.
Proposed fix #1339 changes current behavior and does not solve the errors in initialization.

---
RX initialization/startup issue resolved:

Rx fail fallback values were wrong during init phase.
Without a (powered) receiver connected they stayed there after init.
With a powered receiver they assumed the correct state after init
(causing possible switch events on startup).

Code optimizations:

Taking expression out of the loop.
Prevent double call to function.
Eliminate functio [... cut]