cleanflight/cleanflight #1346: SPRACINGF3 MSP Telemetry

mosi0815, 2015-09-25:
I have wired my MinimOSD to UART1. 115200 baud rate. It works until I arm the board. The motors start spinning for the fraction of a second and then everything locks up. Both LEDs are constantly on and the OSD starts complaining about no data. If I connect it to UART2 then everything works fine. Maybe there's something wrong with the code which allows CLI/telemetry sharing. 

I'm on the latest commit:
0d6d83925f6fa5e5f7c2ef1bc5afad1721ef079e

joshuabardwell, 2015-09-25:
Chances are that you have something else on UART1, such as Blackbox or telemetry, that is taking over the UART once you arm.

mosi0815, 2015-09-25:
[image]

For me this looks right.

joshuabardwell, 2015-09-25:
You have MSP telemetry enabled on UART1. You don't need that. It's not the same as MSP data. Because you have telemetry enabled on UART1, as soon as you arm, the OSD stops working. The OSD uses MSP data, not MSP telemetry.

mosi0815, 2015-09-25:
Ok. Now it works. Thank you. But isn't it kind of weird that the FC crashes completely if its configured like that?

joshuabardwell, 2015-09-25:
Maybe you have found a bug after all!

hydra, 2015-09-26:
Does it actually crash?  One LED would be on when armed and the power LED is always on.  If you disarm does the LED go off?

mosi0815, 2015-09-26:
No. It stops the motors after the fraction of a second and doesn't react on RC input anymore. (SBUS)

hydra, 2015-09-27:
Thanks for confirmation. We'll investigate ASAP.

hydra, 2015-09-28:
Unmarking as critical since it's not a safety or flight issue.

hydra, 2015-09-29:
Removing milestone, we'll fix when we can.

eziosoft, 2015-10-13:
Actually I think that there is something wrong with MSP handling. I was able to put the board into strange state where it stopped to react to MSP.
I didn't have a time to test it more.

Board was dissarmed. I was using EZ-GUI through 3DR Radio (57600). 
Board stoped responding to MSP and what was really strange one motor rotated slowly few times.
I was able to reproduce this behavior quite easily.
Board: sp3

joshuabardwell, 2015-10-13:
This may or may not be related, but I'll toss it in: I have occasional hard locks on the board after making PID adjustments via MWOSD and transmitter sticks. I hit "save", and the settings are saved, but then all three LEDs are lit and the board is completely unresponsive. MWOSD shows "No Data".

I haven't reported it until now because I can't reproduce it with any reliability, but perhaps it is related.

Flying BetaFlight on a Naze32 v5 with MSP on SoftSerial at 9600 baud.

Netsplits, 2015-11-07:
@joshuabardwell you're not alone. I reported lockups on FC and can reproduce every time when using MWOSD on softserial with a GPS on the other softserial. I opened issue #1249 where I explain my problem that is probably related to this one.

Also can confirm having occasional lockups when tunning PIDs with MWOSD.

Netsplits, 2015-11-07:
@eziosoft in your test, I understand that you are using a hardware serial. Is it shared with another function?

eziosoft, 2015-11-08:
Yes I use hardware serial and it is not shared with anything else. At the moment I have connected 3Dr radio to uart2 and sbus receiver and it still hangs from time to time.

eziosoft, 2015-11-08:
Btw when the board stops responding to MSP I'm also unable to arm it. (I use sbus)

closed 2017-02-25 by hydra