cleanflight/cleanflight #1347: Naze32 1 Softserial and LED_STRIP and CURRENT_METER

bitogre, 2015-09-25:
Currently, if you enable SOFTSERIAL, you can no longer use LED_STRIP and CURRENT_METER.  This look like it is because enabling SoftSerial enable 2 ports.  But what if I only need one?  Ideally, I'd like to use only 1 softserial port TX only for FrSky Telemetry and still use LED_STRIP and CURRENT_METER.  Looking at the pinout, I would think this should be possible but that is only based on a minimal investigation.  Is there a reason why this is not possible on the Naze32?

Roosted7, 2015-09-25:
Yes, there is another reason why these function don't work together besides the pinout.

Softserial, led_strip and current_meter all use the same internal timer in the F1 processor on the Naze32.
So because softserial and led_strip both need a timer with different settings, and there is only 1 'free' timer left, both function can't operate at the same time.

A F3 based board (like the SP Racing F3) has enough timers to run all these functions at the same time.

hydra, 2015-09-26:
Duplicate of #465 - Please search!

closed 2015-09-26 by hydra