cleanflight/cleanflight #1348: CF steers back after stick centered in Level modes

Thor368, 2015-09-25:
Hi to all,
I noticed a very strange behavior I can't get my head around. Frist up my setup: I am flyging a Afro PRO on an Q450 with either 8" or 12" Props. Motors are 950kV @ 3s. The quad flys OK with my 8" Props on. I did very little tuning to the Harakiri PIDC (4.0/0.04/23 P/I/D) at Roll and Pitch. Yaw is at standard 8.5/0.045/0. Level P is at 9.0. The problem with my current 8" Props is too low thrust to do anything but hovering. So I got myself 12".

That is where the Problem starts. Obviously the PIDs need to be reduced regarding the new extra thrust. I did this by lowering Level P to 5.0 and Roll/Pitch P to 2.0. This made the quad at least flyable. But as I did some maneuvers and build up some speed I noticed that the quad steers back after releasing the stick to center?!?

Imagine this: You do fast forward flight and then promptly release the stick. The quad should level itself out and slowly decrease speed due to drag. But what happens is CF steers back as if it wants to break by itself. This happens in every direction and more freakishly exactly to zero translative movment from any speed! When zero speed is reached the quad rapidly tilts back to level. Right on spot. It's like magic!

To track this down I nulled out I and D for Roll and Pitch complety and reduced Level P to 3.0. Besides that the quad was almost unflyable with Level P so low the problem stayed exacly the same. With the 8" Props none of the above happens. At least it is not noticable.

Has anyone an idea what this might be and what I could try to fix it?
Thor

joshuabardwell, 2015-09-25:
This is a support question, and belongs on RCGroups or the IRC channel. Github is for feature requests, bug reports, and other development topics.

Post your question here: http://www.rcgroups.com/forums/showthread.php?t=2249574

ledvinap, 2015-09-25:
@Thor368 : this is mainly caused by the way Cleanglight does attitude calculation - it is assumed that acceleration (gravity) points down, but during deceleration (in horizontal plane) resulting acceleration vector (quad acceleration + gravity) is shifted.
This creates difference between 'true down' and 'IMU  down'. Level PID has 'IMU down' as reference input and it aligns quad in this frame (tilting opposite to flight direction). This leads to faster deceleration, stooping quad quite fast.

Thor368, 2015-09-25:
This seems to make sense. I saw the quad tilting way beyond its max angle for level modes. This would only be possible if the IMUs reference frame is tilted. On the other hand the selfstabilisation at zero movement can not be explained in this manner. The quad has no measurment for "speed". If the referenz frame is tilted it should happyly stay so even at the tipping point from forward movement to backward movment.

I already studied CF IMU algorithms. In this case maximising gyro_cmpf_factor should significantly easy the problem. I will report back if it did.

ledvinap, 2015-09-25:
According to literature, there is considerable drag caused by blade flapping, it is speed dependent. Maybe it is necessary to maintain tilted state.. But I am just guessing

hydra, 2015-10-06:
is there anything to do here?

Thor368, 2015-10-06:
Sorry, for my absence. I had a crash recently and destroyed a motor. I could use one of my other quads but these are all miniquads and I never noticed the problem there. The new motors are on the way and the crash was a welcome excuse to upgrade to motors that are more suitable for the big props (as allways;). I will report back when I fixed my craft and did some more tesing.

Thor

closed 2017-02-25 by hydra