cleanflight/cleanflight #1349: Feature Request: Send Buzzer State, Arm State, Flight Mode over Telemetry

AustinSaintAubin, 2015-09-26:
I was wondering if it would be possible to have the flight controller send its buzzer state (1/0) over telemetry.

I currently have my radio receiver analog pin set to a voltage divider on the buzzer pin, which is shared by the buzzer. Doing this allows me to have my transmitter controller admits the tone whenever the buzzer sounds. I would like to remove the wired mess and use my analog pen for other things on my receiver, since it should be possible to have the flight controller send its buzzer state over telemetry.
I am using a Frsky Tranus+, Naze, frsky D4R-II.

--
### Feature Proposal
Send buzzer state, flight mode, and arm state over telemetry.
Not sure the best ID to send the buzzer state over, there are some open ID's and some predefined ones on: [github.com/opentx/opentx/blob/master/radio/src/telemetry/frsky.h](https://github.com/opentx/opentx/blob/master/radio/src/telemetry/frsky.h)

- Add Telemetry Feature:
	- buzzer state over telemetry (1/0).
	- armed state over telemetry (1/0).
	- flight-mode over telemetry ([ARM, ANGLE, HORIZON, BARO, ..](https://github.com/cleanflight/cleanflight/blob/master/docs/Modes.md))
		- A 32 bit unsigned integer for the enable modes might be the best option for sending the enabled flight modes. Each bit in its respective place could represent its mode.
		- The [Mode ID's](https://github.com/cleanflight/cleanflight/blob/master/docs/Modes.md) could represent their bits place in the 32 bit unsigned integer.
		- Would have to work with Open-TX dev team to implement binary notation for Logical Switches in [Open-TX](https://github.com/opentx/opentx) for Firmware & Compainion.
- Add CLI commands: 
	- enable/disable buzzer state over telemetry.
	- enable/disable armed state over telemetry.
	- enable/disable flight-mode over telemetry.

dkisselev, 2015-09-26:
Which specific open and predefined ID's do you see for the features you've mentioned?

hydra, 2015-09-28:
Duplicate of #1297

closed 2015-09-28 by hydra