cleanflight/cleanflight #1354: SP Racing lockup when enabling telemetry

watak, 2015-09-28:
I have a genuine SP Racing with Betaflight cleanflight 1.10 RC2.
Running RX_SERIAL - SBUS  on UART3 and MSP Telemetry on UART1 if I enable Telemtry in Configuration the board freezes up every time I try and arm.

I see the same issue was found in an earlier version:

https://github.com/cleanflight/cleanflight/issues/50

Thanks.

SteveAmor, 2015-09-28:
Is this a duplicate of #1346 ?

watak, 2015-09-28:
Not sure it's exactly the same, telemetry still works when "telemetry" feature is disabled in the configuration.

here is my Ports config.

[image]

If the board is armed then the motors briefly spinup and then it locks up (solid red and blue light). Nothing to do but cycle power.

watak, 2015-09-28:
Well disabling telemetry on UART1 but enabling it as a config solved the issue, sorry about the duplicate in this case.

hydra, 2015-10-02:
duplicate of #1346 - search before raising new issues please.

closed 2015-10-02 by hydra