cleanflight/cleanflight #1355: Interesting stabilization issue

PaulHilgeman, 2015-09-28:
I have had this happen 2-3 times now and finally got it on tape WITH flight recorder data!  I can not re-create this.  It happened once while windy (recorded here) and has happened at least one other time when not windy.  It has happened with pitch both times and always was pitching rearward.  The quad is in Angle mode, and I have not had the wherewithall to switch to Acro or Angle while this is happening yet, I usually just throttle down and crash.

Here is the flight recorder data:

https://www.youtube.com/watch?v=sgMjPXwMlR4

Here is the on-board camera video:

https://www.youtube.com/watch?v=LCsyurlJGQI&feature=youtu.be

I can dump my flight settings when I get home, so stay tuned.

digitalentity, 2015-09-29:
Can you upload your blackbox log instead of a video?

PaulHilgeman, 2015-09-29:
Hmmm, I probably could, but it seems like I cant drag and drop it here, can you reccomend a service?

digitalentity, 2015-09-29:
@PaulHilgeman you can rename a LOG from txt to, say, jpg and upload it here. Or you can use any file sharing service you like - Dropbox, Google Drive.

PaulHilgeman, 2015-09-29:
[image]

digitalentity, 2015-09-29:
What version of Cleanflight is this? Also can you share some details about your setup (flight controller, additional sensors)?

PaulHilgeman, 2015-09-29:
It is a basic 250mm CF kit, Blade 250 by diatone
Emax 1806 motors with 6045 props
12A Spider ESCs – SimonK Firmware
3S battery
CC3D controller running 1.9.0 and Configuaretor 0.65
No additional sensors
500g or so

This e-mail may contain privileged or confidential information. If you are not the intended recipient: (1) you may not disclose, use, distribute, copy or rely upon this message or attachment(s); and (2) please notify the sender by reply e-mail, and then delete this message and its attachment(s). Underwriters Laboratories Inc. and its affiliates disclaim all liability for any errors, omissions, corruption or virus in this message or any attachments.

ledvinap, 2015-09-29:
@PaulHilgeman : the loq is unusable at this rate. Is there any reason to set blackbox_rate_denom higher than 1?

PaulHilgeman, 2015-09-29:
It was for space concerns.  I will try to recapture with a lower denom, I only have 2MB internal flash

This e-mail may contain privileged or confidential information. If you are not the intended recipient: (1) you may not disclose, use, distribute, copy or rely upon this message or attachment(s); and (2) please notify the sender by reply e-mail, and then delete this message and its attachment(s). Underwriters Laboratories Inc. and its affiliates disclaim all liability for any errors, omissions, corruption or virus in this message or any attachments.

ledvinap, 2015-09-29:
Just guessing, but at least 5x quad natural frequency of quad will be necessary. That is about 50Hz, blackbox_rate_denom = 5 for 3500us loop time.

closed 2017-02-25 by hydra