cleanflight/cleanflight #1358: Request confirmation of anybody flying X4R-SB with SBUS on 1.10 RC code

joshuabardwell, 2015-09-30:
An ongoing discussion is occurring on the BetaFlight thread where people are reporting X4R-SB receivers running SBUS having signal dropouts between the receiver and the flight controller. This problem continues even after flashing the latest firmware to the X4R (the CPPM firmware from May of this year, which also outputs SBUS). 

As yet, nobody (including myself--sorry) has taken the initiative to flash 1.10 RC to see if the problem exists there, so I am coming at it from the opposite direction. Is anybody here flying 1.10 RC with an X4R-SB and SBUS? The dropout issue is prevalent enough that, if you were having it, you would definitely know.

Sorry to post a new issue without more confirmation that it really is a bug, but if 1.10 were to have this bug and release, it would kind of be a big deal. And if some people here can say, "I'm doing it and have no problems," then we can be reasonably confident that this is a Betaflight issue, and not in 1.10.

borisbstyle, 2015-09-30:
Yup reproduced here with the latest non-EU firmware. I think its a receiver/firmware issue
[image]

joshuabardwell, 2015-09-30:
Okay, thanks. I will close this.

closed 2015-09-30 by joshuabardwell