cleanflight/cleanflight #1360: Some JR DMSS satellites not working

andypnz, 2015-09-30:
Cleanflight currently supports JR XBus (single line protocol), and also the use of the Align JR RJ01 satellite.

By chance, the RJ01 satellite option also works with the following JR satellites:
RA01T (white PCB)
RA01TL (white PCB)
RA01L (white PCB)

However, the option does not work with older black/dark brown PCB satellites with the same model number (RA01T, RA01TL, RA01L). 

More importantly, it does not work with the new diversity satellite RA02TL. This is a shame, as the RA02TL is 'full range' just with the single satellite.

I imagine the issue is just some small checksum or protocol difference? It cant be too great a change, as all these satellites work fine with any JR DMSS receiver......

I can help get required information from JR, and help with testing. Maybe @ledvinap or @GruffyPuffy have time to look at this?

Thank you!

Donykurniawan, 2015-10-15:
with align RJ01, throttle detect in AUX1 evan i change in radio type . with JR XG, i cant change device premary(throt,ail,ele,rud) like taranis radio. with RJ01 range to short too.

Donykurniawan, 2015-10-15:
problem else with RJ01, when i want use ing rj01 with gps in naze32, 1wire xbus connect to pin 4, its same to use gps pin for rx. . any solution for fixs this.

andypnz, 2015-10-15:
For JR DMSS satellites, assigning the channels as AER2T134 should work, but do check. Remember to click save after changing the assignment.
-	Click on the ‘CLI’ tab
-	In the box which says ‘Write your command here’, type ‘map AER2T134’ and hit enter
-	Now type ‘save’ and hit enter
-	The FC will now reboot
After 30 seconds or so, click on the ‘Receiver’ tab again and check!

Yes, RJ01 is 'short range'. 

You can use RA01L or RA01TL - both long range (but no diversity - no different than a single Spektrum satellite). But it must be the WHITE PCB version of the satellite - the black/brown PCB version does not work. Also, the new RA02TL does not work.

Regarding GPS - maybe you can connect this to a 'soft serial port'? I am not sure.....

Donykurniawan, 2015-10-21:
thank you andypnz, i will try soon, now my cleanflight have a problems. cant back to ready status aftersave and reboot. no showing grafik in sensor tap. any one know how to fix this.

flypilot, 2015-11-09:
Maybe not the correct place to ask this but how does one hook up the 4 wires to the RA01 series receivers? It seems to have two signal wires, which ones goes where? Any help i appreciated..Thanks

Redshifft, 2016-03-02:
Look Here http://www.rcgroups.com/forums/showthread.php?t=2018101&page=4

Is it normal for Cleanflight GUI to get Stuttery in the RC monitor screen ?

closed 2017-02-25 by hydra