cleanflight/cleanflight #1366: Low pass filtering and general filtering documentation

varoudis, 2015-10-02:
With the latest commits, documentationfor the new filtering code my @borisbstyle an others, is ether outdated or not existent.

@borisbstyle maybe you've got something in the works?

Thanks

hydra, 2015-10-02:
In the mean time there might be some useful info from the rcgroups threads and also here:

http://www.rcgroups.com/forums/showthread.php?t=2439428

borisbstyle, 2015-10-02:
I thought @joshuabardwell did some documentation for this?

here a graphical overview of 1.10 filters.
[image]

varoudis, 2015-10-02:
Anything regarding proposed "default" settings? or recommended settings to start with?
Thanks @borisbstyle for all the good work. And @hydra of course.

borisbstyle, 2015-10-02:
- gyro_lpf 188 (lower on very noisy frames)
- looptime 1000 on f3 boards or 1500 on f1 boards
-gyro_cut_hz 60
- dterm_cut_hz 17

If you really have a lot of micro vibrations on P term you might enable
that one as well, but I expect above should be fine.

varoudis, 2015-10-02:
thanks a lot!

1500 loop on a F1? I currently have 3500 on the naze32.. how do I get a good 1500us loop from that CPU?!

borisbstyle, 2015-10-02:
Just change it to something lower. 1500 is usually nice and stable value on
naze32.

varoudis, 2015-10-02:
I had the impression that the F1 cpu is not good for anything lower than 3500 :)
Thanks a lot

joshuabardwell, 2015-10-04:
There is more discussion of the soft filters in [this thread](http://www.rcgroups.com/forums/showthread.php?t=2484202) than in the "Mini quad tuning" thread, which focuses more on PID tuning than soft filters. @hydra thanks for the shout-out.

Tuning the soft filters without Blackbox is pretty difficult, I think. It's worth keeping in mind that the d term cut filter alone will vastly improve the flight characteristics of a copter, because it will allow raising D to a level where it can properly do its job, without introducing excessive noise. Without dterm cut filter, if you raise D past a certain point, you get bad flight characteristics from noise amplification. So one approach for somebody who didn't want to dive in wholly would be to keep gyro_lpf=42 (the default) and add dterm_cut_hz=17 or thereabouts, then raise D gain to see if reduction in bounceback can be achieved. Increased P may also be possible.

Thor368, 2015-10-23:
Hey guys! I had some time and found it beneficial to support this great project with it. Maybe this general overhaul could be usefull for this issue?

closed 2017-02-25 by hydra