cleanflight/cleanflight #1368: Afromini Amaze rev3 (Mini Naze32) Cleanflight Support

AustinSaintAubin, 2015-10-03:
Can we work towards getting full clean flight support for [Afromini Amaze rev3 (Mini Naze32)](http://www.getfpv.com/afromini-amaze-rev3-mini-naze32.html).
- Barometer is not working, see #1232
- Channel order mapping seems to be confused as of Cleanflight v1.10.0
  - Channel mapping was working fine with Cleanflight v1.9.0.
  - Tested with Airplane Mixer, all defaults.

jcowgar, 2015-10-03:
Does the include the beeper or does the board have an error on it preventing the beeper from working?

lazd, 2015-10-08:
Looks like #1326 included support for the BMP280 barometer, so you can scratch that off the list.

xoxota99, 2015-10-18:
From the [discussion on RCGroups](http://www.rcgroups.com/forums/showthread.php?t=2163400&page=65), it seems like even Baseflight doesn't properly support the buzzer on Afro Mini.

closed 2017-02-25 by hydra