cleanflight/cleanflight #1369: Feature request: Blackbox download via cleanflight.

jaimevisser, 2015-10-03:
If the Openlogger is connected to uart2 on a Naze with both TX and RX it becomes possible to dump a file to serial. With a small change in cleanflight it should be possible to redirect this dump to the CLI, allowing you to dump logging.

Rationale: not every OpenLogger is easily accessible inside the quad frame, the USB port on the FC usually is.

Possible snag: is serial fast enough to allow logging download at a reasonable pace?

MJ666, 2015-10-04:
Reading the 16MByte SPI flash already takes ages. don't ting this is a valid option for the OpenLog and is really worth the effort. much better to take the easy access to the SDCard into account during the build phase.

joshuabardwell, 2015-10-05:
>Possible snag: is serial fast enough to allow logging download at a reasonable pace?

In short: no.

MJ666, 2015-10-05:
At least it will not be faster as OpenLog can write. Even for this additional action need to be taken to use an higher baudrate for downloading.

closed 2017-02-25 by hydra