cleanflight/cleanflight #1371: When can support the F4 board?The F4 MCU is much stronger than F1……

Linjieqiang, 2015-10-04:


dkisselev, 2015-10-04:
Right now CF is just starting to support F3 boards, and a few of them have come out specifically for CF.

At this point, F3 boards already complete the control loop faster than the gyro and other sensors can detect changes and get them out of the bus, so there wouldn't be any benefit in a faster CPU since it'll just recalculate based on the same data.

F4 will make sense once we start to see the RAM and Flash filling up on the F3 boards due to new features, but we just switched up to those (and still support F1), so the benefit just isn't there.

joshuabardwell, 2015-10-05:
F4 makes sense on boards where the MCU is connected via SPI. The MCU can update at 8 kHz, but the i2c bus used on many boards can't go that fast.

As Boris has demonstrated, even an F1 board can get sub-1000 looptimes with accelerometer enabled. At that point, the i2c bus is the limit.

hydra, 2015-10-08:
See https://github.com/cleanflight/cleanflight/milestones/STM%20F4%20Targets

hydra, 2015-10-08:
Please do not use the issue tracker in this way, the issue tracker is for bug reports and feature implementation discussions.  The question is too vague anyway, no description was even provided as to WHY we might want to do this (the reasons are already known anyway).

closed 2015-10-08 by hydra