cleanflight/cleanflight #1372: Improve Mag calibration

HaukeRa, 2015-10-05:
The Magnetometer calibration relies on the rotation around all axes. It's easy to miss a needed point during calibration. 

There is an application note which fits a sphere to data points, and using its center as offset values. 

It is way more robust than the current solution as it does not require a specific rotation pattern. 

I already implemented this code for a different project, where I'm not allowed to copy it from, but I would help reimplement it.

The method involves solving a 4x4 matrix but is formulated to use constant memory. (4x4 matrix and two 4x1 vectors)

digitalentity, 2015-10-05:
@HaukeRa
That's perfect news. Corrent way of calibrating a magnetometer is very tricky. If your solution makes that easier - please do it. @hydra, what do you think?

ledvinap, 2015-10-05:
It may be better to handle mag corrections in configurator and just set correction factors in CF ..

HaukeRa, 2015-10-05:
Since the sensor is susceptible to influences of external magnet fields which permanently alters the offset it is really recommended to do it in cleanflight directly.

digitalentity, 2015-10-10:
@HaukeRa, can you direct me to this appnote you referred to? Compass calibration is key to my new navigation system and current calibration method does not always give good results.

HaukeRa, 2015-10-12:
Im talking about  Freescale AN4246

I will reimplement this until the weekend hopefully. As I'm new to CF development it will take some codestyle changes I think. 

I use malloc'ed memory to store the matrix, I do not yet know if CF is fine with this.

digitalentity, 2015-10-12:
There is no malloc on cleanflight. You'll have to statically allocate the
memory. Looking forward to see your implementation.

HaukeRa, 2015-10-12:
I feared that, and hoped for it also :D

digitalentity, 2015-10-13:
@HaukeRa 
It just occured to me that magnetometer can not only be offset by some value, but also its different axis can be scaled differently. Current cleanflight code does not account for this, only offset is compensated for. Do you think it's worth modifying the code to compensate for different scale factors as well?

ledvinap, 2015-10-13:
@digitalentity:
IMO Freescale AN4246 contains math to calculate soft iron influence, but then simplifies the problem to offset calculation (and abs(B)) only. So gain influence is probably not that bad.

Axis scaling may be improved by using calibration data  on AK8975 (ASAX, ASAY, ASAZ Sensitivity Adjustment values). https://github.com/ledvinap/cleanflight/blob/local/src/main/drivers/compass_ak8975.c#L96

Self-test mode may be used to compensate temperature dependence (this is documented only for HMC5883L, but may work for AK8975 too)

digitalentity, 2015-10-15:
@HaukeRa I recently stumbled upon Bill Premerlani paper on automatic magnetometer offsetting here https://drive.google.com/file/d/0ByvTkVQo3tqXY3JFSzVkR1pNNnc/view 
Seems very easy to implement. Probably current mag calibration algorythm can be replaced by this one?

HaukeRa, 2015-10-15:
I'll have a look on this also, looks extremely simple! But the more algorithms available, the better!

digitalentity, 2015-10-15:
Just thinking here. When calibration is started, current code zeroes out bias vector and calculates it after 30 second data collection. 
But if we will use Premerlani's approach, we can start with any initial estimate of bias and don't zero bias vector when calibration starts. This way we can have incremental calibration. 
Imagine - if the calibration didn't converge in 30 seconds - repeat and bias values will be *refined* using previous result as initial estimate.

HaukeRa, 2015-10-15:
My current implementation collects a predefined number of samples which are at least some degrees (i.e. 7.5 degrees) apart. It showed that this is enough too ensure a good calibration.

digitalentity, 2015-10-15:
How about me implementing Premerlani's approach and you implement yours and compare the results? Lets find the best way :-D

HaukeRa, 2015-10-15:
Totally fine with that! :D

digitalentity, 2015-10-15:
I've written some code (see f87f8c186fd9d9c8ffd5faca6735071407673b13). It will require some fine tuning, but this can wait until tomorrow - time to sleep now.

ledvinap, 2015-10-15:
@digitalentity : Premerlani approach seems to be dependent on quite large attitude difference between  each iteration. With small changes, it will be integrating noise. Rounding errors may cause significant problems..

digitalentity, 2015-10-15:
That's why I do not intend to use it inflight. Manual mag calibration
introduces required large changes to attitude and the code will use only
readings that are significantly different from each other. Today's testing
will show if it works.

HaukeRa, 2015-10-16:
Implemented my approach today, but need to catch some sleep now. Need to test it tomorrow,  and benchmark speed. Also memory footprint is not as bad as thought.

digitalentity, 2015-10-16:
@HaukeRa 
I've tested my code today. Calibration converges to sane offsets very fast. Result is very consistent and very easy to achieve - activate calibration and wave the copter around the the air, no need to follow the exact 3-axis pattern. In fact I'm surprised how well it works for me. Bill Premerlani, you are a genius!

digitalentity, 2015-10-16:
After some tweaking and bugfixing here is the final code - 2fb43b4484046d74802db7a65009014d09dfbbb0

I deliberatly set magZero to 2000 on all axis and new algorithm was able to converge on good offset values in 3 iterations (3 times pressing "Calibrate mag" button and waveing the copter). Each iteration was giving better results.

I tried to set gain to 0.5 or even 1, but result was not very stable. Gain of 0.05 gives a good balance between speed and accuracy.

@HaukeRa, can't wait to see your code and try it on my NAZE32 full. Is your code dependant on following a certain rotation pattern or I can simply wave the copter in the air?

HaukeRa, 2015-10-16:
First implementation, the linear equations code should go elsewhere  etc. but I do not yet have setup an IDE, and am not too familiar with the CF code yet. https://github.com/HaukeRa/cleanflight/commit/09f58b8a55fbaa76c0aa4aa2af87e26f52eb88c6

Tested it once here, and it seems to work, but that was just one quick test.

Currently it takes every sample until 128 samples are collected (takes about 12.8 secs at 10Hz). To improve this I need access to the IMU and didn't know how to yet.

@digitalentity Just do some movement, you don't need to go full 360°, but the more you cover each axis the better the result will be. Doing an "eight-shaped" wave has proven to be pretty easy and sufficient.

digitalentity, 2015-10-16:
Looks good. Will test tomorrow, don't have a naze32 at home. I have a few remarks though:

IMU will be unreliable when mag is not calibrated. Testing if mag readings are significantly different from each other should be enough (probably you could calculate a difference vector between current and previous readings and calculate magnitude of that vector).

Your code works too fast. 12 seconds might not be enough to rotate a copter significantly. Also tools like configurator assume that calibration takes 30 seconds and there is no flag to indicate calibration finish. Can your code work with variable number of samples? I.e. Collect samples until 30 seconds is over and calculate bias after that.

HaukeRa, 2015-10-16:
Its no problem to scale the algorithm to any amount of datapoints, as long as the floats are sufficient to hold the values.

So 30 Seconds is no problem, at worst one would stop to accept samples from some point in time on.

digitalentity, 2015-10-16:
Have a look at acda277 - I've made a few changes to your code - restored 30 sec timeout, added a condition for accepting new samples only if significantly different from the last one, made calibration independent from align_mag.

HaukeRa, 2015-10-16:
Idk if the 50*50 distance ist right, or how much rotation that represents but in general it looks good.

digitalentity, 2015-10-16:
Distance of 50 is a wild guess that worked in my code. Premerlani's approach is very succeptible to noise so I had to use only significantly different readings. When testing it looks like distance of 50 equals about 10-15 deg of rotation.

HaukeRa, 2015-10-16:
In my other project I used 8 degrees, that resulted in a good balance between the number of points and a distribution over all possible points.

digitalentity, 2015-10-16:
Have a look at 0a6b20a - does this looks reasonable?

ledvinap, 2015-10-16:
it is probably bad idea to filter point based on distance - outliers caused by noise are likely to be selected by this filter (if rotation is slow compared to sampling, outliers are the first one to cross threshold)

digitalentity, 2015-10-16:
@ledvinap 
I also hate the possibility of noise messing up with calculated offsets, but without access to IMU we'd not be able to reilably calculate angular distance between two measurements. 
And if we get IMU involved, or throw some serious math at this, the code might get too complicated. Having compass offsets correct to 0.05% accuracy (LSB of a typical mag chip) is an overkill. 
We want calibration to be reasonably precise, but we don't want to spend lots of memory on a feature that would be used only once.
Maybe the code will provide reasonable results the way it is. Or maybe a simple 3 or 5-point median filter will to the trick?

HaukeRa, 2015-10-16:
From my experience noise is not an issue during calibration when motors are not spinning. 

Can't we run the IMU without mag influence during calibration? That's the way I'm doing it.

ledvinap, 2015-10-16:
@digitalentity : with Freescale / HaukeRa algorithm it is not necessary to discard samples - just use all you get.

digitalentity, 2015-10-17:
@ledvinap, you are probably right given that our accumulator variables won't overflow.

ledvinap, 2015-10-17:
floats will be fine .. underflow (rounding) is bigger problem if most of the samples are from one pose.

digitalentity, 2015-10-17:
That's why we probably need to make samples somewhat different from each other.

ledvinap, 2015-10-17:
Samples that define 'good' calibration matrix are needed.
If you rotate quad in 6 directions (+-X, +-Y, +-Z) and collect (about the same number of) samples in each pose, the matrix will be almost perfect. Collecting more samples without moving will reduce noise in measurement.

The problem may appear if you collect almost all samples in X+ direction and then only few in others.

digitalentity, 2015-10-19:
I have some comparative test results.

Platform: NAZE32 Full on a long usb cable, in an environment mostly free from strong magnetic interference (no magnets, no iron, no power wires in close proximity).
Code used modified @HaureRa's (f4e7f09663ae8eac57d49a988359e5320d179b25) and @digitalentity's (e0c90808ec83150d3d170980243feb9da121f209) algorythms. Both algorythms were rebased on the same commit 526f4777a24e6963ded6e6b92060350aa0e95878.

Measured MAG offsets: (-82; -297; -80)
All testing was done by zeroing out offsets, starting mag calibration and making random movements covering 360deg rotation on all 3 axis. Calibration repeated three times, every result recorded.

Results for @HaukeRa's code:
Footprint: flash=121236, bss=14072
Results: (-89, -300, -79), (-102, -286, -74), (-81, -296, -67)
Sigma: (10, 7, 6)

Results for @digitalentity's code: 
Footprint: flash=120748, bss=13992
Results: (-77, -301, -58), (-90, -312, -73), (-87, -320, -90)
Sigma: (6, 6, 18)

Both algorithms provided stable and usable results, close to measured bias vector in all 6 cases with almost zero effort. Heading error when tilting calibrated copter was about +-5-10deg.

digitalentity, 2015-10-19:
Looks like your code gives slightly more consistent results at the cost of 488 bytes larger footprint.

digitalentity, 2015-10-19:
Another test of @HaukeRa's code (this time without my modifitations to avoid consuming "close" results ()

Footprint: flash=121148, bss=14056
Results: (-88, -296, -74), (-113, -292, -65), (-124, -279, -103)
Sigma: (18, 10, 20)

Much less consistent results. The latter is unusable, 
Looks like keeping results significantly apart is very important to

digitalentity, 2015-10-19:
And finally a test with my code with gain lowered from 0.05 to 0.02 and performing calibration incrementally (without zeroing out previous results)

Results: (-74, -231, -59), (-87, -285, -76), (-82, -309, -85), (-87, -308, -86)

Algorithm was unable to converge on the first try, but on the second, third and forth try results were consistent and gave a very good, almost drift free heading.

HaukeRa, 2015-10-19:
I'll do some testing of my reference implementation today, and calculate sigma of a few calibrations.

ledvinap, 2015-10-19:
@digitalentity : Can you try collecting mag samples for offline analysis? Storing them in debug[] and dumping in configurator should be enough.
It would be interesting to compare algorithms on identical data / visualize what's happening

HaukeRa, 2015-10-19:
I can dump some data from my platform. If I'm right I might also still have the algorithm as Matlab code with visualization

ledvinap, 2015-10-19:
@HaukeRa : That would be great.

HaukeRa, 2015-10-19:
Preliminary Results, full data will follow later:

Finished Mag calib: [-105.491661,104.455223,-53.381275]<\n>
Magnetic field strength: 53.140007<\n>

Finished Mag calib: [-105.053452,105.288170,-54.950687]<\n>
Magnetic field strength: 52.492058<\n>

Finished Mag calib: [-104.632370,104.323296,-53.791111]<\n>
Magnetic field strength: 52.619766<\n>

Finished Mag calib: [-106.943138,103.560295,-52.033485]<\n>
Magnetic field strength: 52.990677<\n>

Finished Mag calib: [-103.419296,105.593552,-53.504028]<\n>
Magnetic field strength: 52.922459<\n>

Finished Mag calib: [-106.868736,102.695015,-52.921749]<\n>
Magnetic field strength: 52.298885<\n>

Finished Mag calib: [-104.528847,104.413376,-53.157269]<\n>
Magnetic field strength: 52.217995<\n>

Finished Mag calib: [-103.570480,105.902817,-52.655590]<\n>
Magnetic field strength: 52.160889<\n>

Finished Mag calib: [-106.554375,104.374329,-52.866940]<\n>
Magnetic field strength: 52.732746<\n>

Finished Mag calib: [-104.496002,102.057251,-54.522366]<\n>
Magnetic field strength: 51.343071<\n>

Sigma^2 = (1.6488, 1.4893, 0.7575)

Sigma = (1.2841, 1.2204, 0.8703)

Taken 64 samples each try. Samples were chosen will be denoted in the logs.

HaukeRa, 2015-10-20:
The log: https://drive.google.com/file/d/0BxJvibam28UTWUpjSW02b0FvS01XRzBPZ2NtSXJVaDVFTDdn/view?usp=docslist_api

#MI; Sample is ignored for calibration
#MC; sample used for calibration

HaukeRa, 2015-10-21:
Im wondering why I'm getting way more consistent results. The only differences in software are, that I use a different algorithm to select the samples based on the IMU and I scale the mag values to uT before pushing them into the matrix. 

@digitalentity

digitalentity, 2015-10-21:
How do you test? Are you following a specific rotation pattern? Maybe you
have less external magnetic fields.

HaukeRa, 2015-10-21:
Didn't follow any precise pattern, but try to achieve  360 deg on all axes. Feel free to compare them to your readings!

digitalentity, 2015-10-22:
I've re-tested the code in a more compass-friendly environment (in the field actually) and your code gave amost the same results three times. (Calibration with random movements, trying to cover all directions). My code performed slightly worse but it was unnoticable on heading (probably gave less than 1-2 deg error)
Don't have any numbers - used a piece of paper to write everything down and it eventually got blown away by the wind :angry:
Looks like we both have algorythms performing equally well in real world. I've merged my code to my nav-rewrite branch to let people test it and very positive reports are coming in.

digitalentity, 2015-10-25:
@HaukeRa 
I wonder if the same approach can be used to calculate accelerometer offsets. Collect average readings at 6 stationary positions (not necessary orthogonal) and use the very same algorithm to calculate offsets. What do you think?

digitalentity, 2015-10-25:
Just tried Freescale's approach on accelerometer - excelent result and surpisingly different from current code. I guess current accelerometer calibration code does not calibrate anything at all.

ledvinap, 2015-10-26:
@HaukeRa : I did analyze the data you posted.
Using http://www.mathworks.com/matlabcentral/fileexchange/45356-fitting-quadratic-curves-and-surfaces.
ellipsoidfit_direct : 
```
center = [ -105.811   103.948   -53.246]
axes = [53.280 52.723 50.980]
```
spherical fit (from  Freescale AN4246):
```
center =  [ -105.629   104.235 -53.293 ]
```

Axis sensitivity error / soft iron influence is quite small (5%) and ignoring it causes small offset error (<1%). 

Strange thing is that when I normalize data to sphere with unit radius,  there is some error that seems unrelated to orientation:
(normalized)magnitude:
[image]

scatter plot of magnitude in spherical coordinates (measurement is rotated to match ellipsoid axes):
[image]

Maybe the error is dependent on movement speed. Or there was something causing error according to position ... Any idea?

Code used to analyze data. It is quite ugly, but seems to work:
```
% read data converted to csv
mag = csvread('mag_calib_f.txt');

% fit ellipsoid using direct least squares
e1 = ellipsoidfit_direct (mag(:,1), mag(:,2), mag(:,3));
[center,ax,quat,R]=ellipsoid_im2ex(e1);
% remove offset and scale data to unit sphere
magn=(mag-repmat(center', rows(mag), 1))*R'*diag(1./ax)*R;

% convert to spherical coordinates and plot
[th,ph,r]=cart2sph(magn)
scatter(th,ph,[],r);
% plot magnitude
plot(r);

% Freescale AN4246, no optimization
Y = sum(mag.^2, 2);
X = [mag ones(rows(mag),1)];
beta = (X'*X)^-1*X'*Y;
center2 = beta(1:3)/2
```

HaukeRa, 2015-11-07:
Sorry I was on vacation. 

The error seems strange, I will try to do gather data with my naze to be sure if it is the same problem here.

closed 2017-02-25 by hydra