cleanflight/cleanflight #1376: Naze32 rev6 issues

borisbstyle, 2015-10-07:
I saw naze32 rev6 code has been added to the master, but I noticed 2 things.

The MPU_DATA_READY interrupt code doesnt seem to work and set acc_hardware = 1 also disables gyro it seems.

I dont have the board, but someone else helped me on distance.

Just to report it here.

borisbstyle, 2015-10-08:
@hydra 
Is the interrupt pin same like on rev5 on this one?

hydra, 2015-10-08:
likely there's still some additional testing and fixes to do.   buzz the pin out from the MPU to the CPU to check which pin it's connected to.  Once you confirm it I can test here too, i thought it was the same but didn't check.

borisbstyle, 2015-10-08:
Unfortunatelly I dont have this board yet to test with it. I rely on others feedback for now, but in the gyro sync code as it is I dont get interrupt on NAZE32 rev6 and also not on COLIBRI_RACE target. Both use MPU6500, but COLIBRI_RACE has no exti pins defined.

MJ666, 2015-10-08:
Reading the BF code the interrupt pin schold be the same for all later revisions starting from Rev5.

quadquestions, 2015-10-10:
@borisbstyle Hi Boris, I can send you a board if you need.  just email me, anthony@quadquestions.com

Xcalllibur, 2015-10-15:
I have been monitoring this issue since it was opened and haven't seen any movement.  I have a rev 6 and am keen to help fix this issue.  If someone can give me some guidance on how to proceed I can do the work.  I see above that you suggested to check the pinout between the MPU and CPU, but I am not sure what to do with that information?

borisbstyle, 2015-10-15:
@Xcalllibur That's not relevant now as CF doesn't use interrupt at all. The pinning should be fine according to some people.

@hydra Did you have time to check the code for interrupt and why it isn't working?

tracernz, 2015-10-15:
The gyro interrupt pin is the same as rev. 5. I know this because My [quick rev. 6 hax](https://github.com/tracernz/TauLabs/blob/naze-rev6-hax/flight/targets/naze32/fw/pios_board.c#L142) works, [and wouldn't if the EXTI config was wrong](https://github.com/tracernz/TauLabs/blob/naze-rev6-hax/flight/targets/naze32/fw/pios_board.c#L488) (and also I checked that the attitude estimator task runs at the correct frequency ;) ).

borisbstyle, 2015-10-15:
@tracernz 
I believe you and I know its the same. I was just saying that it should be addressed in CF. Even Colibri Race target what alsu uses MPU6500 has no working interrupt yet.

tracernz, 2015-10-15:
@borisbstyle Yeah, was just adding here to confirm. ;) I guess something must be missing in the CF MPU-6500 driver as you say.

borisbstyle, 2015-10-21:
solved

closed 2015-10-21 by borisbstyle