cleanflight/cleanflight #1377: Arm by Bluetooth

rocboronat, 2015-10-07:
Hi! We're developing an Android app to fly the drone, but we've having issues arming the board.

To arm it, we're sending a 2000 value by the AUX1. So, we enabled the arming by the AUX1, setting that more than 1500 should activate it. Well, also the AUX2, AUX3 and AUX4, to give it a try.

But it didn't worked. The same with other apps that are doing the same :·(

We need to do anything more to arm the drone rather than sending the AUX1 value?

Thanks!

varoudis, 2015-10-07:
Send 'virtual' stick movement to simulate the arm position? Check docs directory,i think there is one image

hydra, 2015-10-08:
The method of transmission is irrelevant.  (bluetooth).  It's the communication protocol that is relevant.

You dont say HOW you're sending it or what your configuration is.

Likely the failure to arm is caused by invalid configuration.

You're using RX_MSP right?  Config dump please.

rocboronat, 2015-10-09:
Actually, not sure about the way I'm sending it. But asking for it is probably the key to solve the issue. I'll check RX_MSP! I'm really new to the drones world, and everything is something new for we. Thanks for helping! I'll note here my progress. Hope I arm it and close the issue :)

eziosoft, 2015-10-13:
Arming works fine with RX_MSP.
Done this many times.

rocboronat, 2015-10-14:
@eziosoft I've use your app to check that the engines are working well :+1:

rocboronat, 2015-10-19:
Hi guys,

first of all, thanks for the support. I'm working on the board again. Now, I know what's RX_MSP. Probably saying "MSP RX input" is too poor for a description... :·)

BTW, Now I guess that I'm not using the good Serial Receiver Provider. Or not using the needed Ports.

So... do you know any guide to make the board fly by BT? I've looked for a tutorial, but no luck. Anyone explains what's the configuration to make it fly by BT.

BTW, I'm attaching my configuration. The Port tab is something random... trying to make it work.

[image]

[image]

thenickdude, 2015-10-22:
You still haven't said what data you are sending to Cleanflight. What is the actual MSP message you are sending and with what payload? (`MSP_SET_RAW_RC` hopefully). Do you have an arming switch appropriately configured on the Modes tab?

rocboronat, 2015-10-22:
@thenickdude sorry! Actually I didn't understood the question. And yes, I'm sending the  MSP_SET_RAW_RC message.

BTW, if you want to go deeper, it's the line 177 of https://github.com/Fewlaps/flone-android/blob/b8f4d91c2b9f0bc821695107bb9abc3888be3a27/app/src/main/java/com/fewlaps/flone/service/DroneService.java

thenickdude, 2015-10-22:
Your serialisation code is here, which looks fine:

https://github.com/Fewlaps/flone-android/blob/b8f4d91c2b9f0bc821695107bb9abc3888be3a27/app/src/main/java/com/fewlaps/flone/io/communication/protocol/MultiWii230.java#L657

Are you running the latest Cleanflight development build? You can grab one from here if you don't want to build it yourself:

http://cleanflight.memoryleaks.org/builds/cleanflight_cleanflight/

What rate are you sending MSP messages at? The MSP code needs to receive at least 5 updates per second, otherwise it'll decide that the Tx is out of range and go into Rx loss mode.

You didn't confirm if you added an arming switch or not on the modes tab, that's a requirement if you want to arm by AUX1. You can test your setup by clicking the "Control Sticks" button on the "Receiver" tab on the latest development build of the Configurator (*not* in the Chrome Web Store, must be manually installed as an Unpacked Extension) and using our built-in MSP joysticks:

https://github.com/cleanflight/cleanflight-configurator/tree/development

rocboronat, 2015-11-01:
Hi @thenickdude and the whole team!

First, thanks for helping, and sorry for my late post. I burned an AfroFlight32 in the meantime... The 12v PDB touched the board Good for me! This is the hard way of developing for hardware, hehe

I just runned the last development build in the board. It seems that my issue is still there.

About the AUX1 arming switch, yes; I added it. I'm sending 1000 for disarmed and 2000 for armed, so we chosen this configuration (I think it's OK):

[image]

About the rate of refresh, it's really high. We're launching a refresh request every time we receive an answer. So, it is refreshing as real time as we can. More than 5 updates per second. But good to know about that automatic action! :·)

About the development edition of the Cleanfight Configurator: great! I found the Control Sticks, and tried to arm the board sending a 2000. And... it didn't work. That's great for me to remove the app of the whole issue...

About all the thing: I recorded a video to showcase the issue I'm happening. I hope it helps! :·) https://youtu.be/Cz0-rL1Gf-8

Ah! A last point: I've tested the drone with your app, and I can make the motors spin. So, they work.

Lots of thanks for your support!

rocboronat, 2016-01-12:
Friends! Thanks for your patience. I very sad to say that the issue was in the board. After getting a Flip32 instead of an AfroFlight32... it worked :·)

Please, if you're curious about my development on working with CleanFlight via Bluetooth and Android, don't hesitate to watch https://github.com/Fewlaps/flone-android

Thanks!

closed 2016-01-12 by rocboronat