cleanflight/cleanflight #1379: VBAT with CC3D: start to beep (buzzer) after repower the board

robustini, 2015-10-08:
When i enable "VBAT" all work fine, when i repower my CC3D the buzzer starts to sound with "low voltage" alarm.
To make it functional again i disable it, enabled again and fly without disconnecting the LiPo.
Please check, thanks!

hydra, 2015-10-08:
please provide better description of the problem.

Please read: https://github.com/cleanflight/cleanflight/blob/master/CONTRIBUTING.md which you should have read (there was a link) before creating this issue.

robustini, 2015-10-08:
You're right, i'm sorry:

1) Configure VBAT (like in my image)
2) Save the setting
3) Fly (tested with 3S LiPo, 12.6V) and wait the voltage alarm (i land with 3.65V for cell, perfect)
4) Change the LiPo with another full charged and now start to beep for "low battery"

[image]

robustini, 2015-10-12:
What i mean is: please, if you possible check the battery cell detection code in "battery.c", dunno if my problem is related to the "stable delay", too short?

#define VBATTERY_STABLE_DELAY 40

ledvinap, 2015-10-12:
The code at https://github.com/cleanflight/cleanflight/blob/master/src/main/sensors/battery.c#L92 is wrong. There is only one sample processed after delay (and no update during delay()).
Value used for cell calculation is invalid ... 

It may be changed into 
```
   for(unsigned i = 0; i < VBATTERY_STABLE_DELAY; i++) {
      delay(1);
      updateBatteryVoltage();
   }
```
as a quick and dirty fix

robustini, 2015-10-12:
Nice shot @ledvinap!

robustini, 2015-10-12:
Imho the "battery cell detection" is too restrictive, if i connet a 3S LiPo and the parameter "voltage scale" is not weel tuned (es, if detect 12.7V) the code detect 4S and start to beep.
It would be useful to have in the GUI the number of cells detected, don't see it.

ledvinap, 2015-10-13:
@robustini :
Max pack voltage for 3 cell is 12.9V (4.3V per cell, vbatmaxcellvoltage) with default settings, higher voltage is considered discharged 4s pack (~3.25V per cell). So there should be 0.1V margin .. 

Maybe this is caused by wrong detection code?

cli command `status` will print detected cell count.

robustini, 2015-10-13:
If you see my image i've put the "max cell voltage" to 4.3, do you think this is the problem?

ledvinap, 2015-10-13:
vbatmaxcellvoltage is used as threshold for cell count detection. Default value has 0.1V margin you want ... 
```
cells = (batteryAdcToVoltage(vbatLatestADC) / batteryConfig->vbatmaxcellvoltage) + 1;
```

BTW: `cells` is based on last ADC reading and not filtered value. My VBATTERY_STABLE_DELAY remark is invalid, but cell count is based on noisy value, which may have very unpredictable results

closed 2015-10-17 by robustini