cleanflight/cleanflight #1381: Automatic ALT Hold kick in

Ravid824, 2015-10-09:
Currently ALT HOLD switch should be turned ON while the the copter hovers. Once you set this mode I still want Throttle control !!!
In the NAZA V2 ALT HOLD (which is the best I experience so far) you don't have such switch.
The ALT HOLD is kicked in automatically once the Copter is in hovering. I think it is kicking into this mode once the Throttle is in the middle point (if you set it properly so that middle throttle will hover). It has about +-5%-10% tolerance around the throttle mid point in which this mode will be kept.
While above or below the mid throttle tolerance user get a full Throttle control (ALT HOLD function should stop automatically).
The advantage of this implementation is that user doesn't need to take care for extra switches during flight and you get much better throttle control.

Thank you guys for the great software. One of the best and stable open source code.
David

ducktaperules, 2015-10-09:
Currently when ALT HOLD is turned on you still have throttle control, raising the throttle stick will increase altitude and lowering the throttle stick will decrease altitude. There is a dead-band to hold throttle defined by alt_hold_deadband, The center is wherever the throttle is when you enable throttle hold.

as far as im aware NAZA does not give you "full Throttle control" outside of the dead-bands, it just adjusts the target altitude in a similar way. But if you do want manual throttle outside of the dead-band then you can also disable alt_hold_fast_change. 

from the code about alt_hold_fast_change:
"When disabled, turn off the alt hold when throttle stick is out of dead-band defined with alt_hold_deadband; when enabled, altitude changes slowly proportional to stick movement"

Ravid824, 2015-10-09:
Thanks you, I will try your advice.
I have enough NAZA time, and i think that I have full control outside of the alt_hold_deadband (The feature is completely turned OFF outside of the deadband). the traditions into and out of the alt_hold_dead-band is very smooth.
And most important, they don't have any switch, the feature turn on/OFF automatically when you are in/out the deadband zone.

closed 2015-10-09 by Ravid824

joshuabardwell, 2015-10-09:
You can easily implement this yourself if you have a reasonably programmable transmitter. Just map your throttle to an Aux channel and then use Modes to select the dead-band range in which you want alt-hold to be active. You will need to carefully select the dead-band range to be wherever your copter normally hovers.

Ravid824, 2015-10-09:
Hi ducktaperules
currently the alt_hold_fast_change default to 1. 
"1" means enabled?