cleanflight/cleanflight #1382: Feature Request: In-flight Adjustments improvments

Ravid824, 2015-10-09:
Currently the following PID controls can be used:
RC Rate, RC Expo, Throttle Expo, Roll & Pitch Rate, Yaw Rate, Pitch+Roll P I and D, Yaw P I and D.
Can you add more PID controls, such as ALT HOLD, POS HOLD and other PID settings?

Also, can you extend the feature and use an Auxiliary potentiometer.
In order to use this feature I need two 3-way AUX switches. One to select the variable I want to adjust and one for Up/Down adjustments. I don't have spare 3-way AUX switches while I would like to utilize these for other functions.
The idea is that when the Pot is in the middle (with +- tolerance) it will not do any adjustments. While turning the POT right outside of the tolerance will do the UP adjustment and left outside of the tolerance will to the down adjustment.
I know that currently I can use the POT for this purpose, but I need to turn it all the way left and right which is not easy to do during flight.

Thank you

joshuabardwell, 2015-10-09:
>I know that currently I can use the POT for this purpose, but I need to turn it all the way left and right which is not easy to do during flight.

You have complete control over what range of values cause an increase/decrease in the adjusted parameter. There's no reason why you should have to turn the pot any further than you want. If you want more explanation on this, post over on RCGroups for a support request and I'll help you.

Ravid824, 2015-10-11:
thank you for support. I sent you a message via the RCGroups.

joshuabardwell, 2015-10-11:
Thanks for the reminder. I read your message and meant to respond when I was in front of my PC, then forgot. I will get back to you today.

joshuabardwell, 2015-10-11:
I have to apologize. I was wrong. The range of values that controls which parameter you are affecting is infinitely adjustable, but the range of values that controls whether you increase/decrease the value is not adjustable. So this is a valid feature request, and again, I apologize for mis-speaking.

The default behavior seems to be:

> high = mid_rc + 200, low = mid_rc - 200. by default this is 1700 and 1300 respectively.

So you have about a 400 step deadband.

Ravid824, 2015-10-11:
200 range to each side is better than full range.
Still wonder, How the code find out if the Aux is assigned with standard switch or POT?
Actually I am asking how to get the 400 deadband activated.

joshuabardwell, 2015-10-11:
The 400 deadband is the default and does not seem to be configurable. The code doesn't care whether Aux is assigned to a switch or a pot. In any case, the channel value goes up and down.

Yamaford, 2015-10-20:
until someone writes this to work like you want can you change the behavior of the pot you want to use in the mixer?

closed 2017-02-25 by hydra