cleanflight/cleanflight #1385: Magnetometer misbehaving when not mounted on the board? ...

AlexanderSidorenko, 2015-10-09:
Hi!
I have Naze32 rev 6. It comes with barometer, but without magnetometer.
I've got myself a GPS+Magnetometer combo (this one: http://amzn.com/B00N3TE312).
The magnetometer is HMC5883L and it is mounted right above the board, but about 30cm higher ...

I did magnetometer calibration countless times and always get the same behavior: magnetometer more or less works when craft is horizontal, but if I roll aircraft to the left 45 degrees, it does yaw rotation precisely 90 degrees to the left... if I roll it to the right, it does yaw rotation same 90 degrees to the right.

It all seems to me as a wrong assumption in math somewhere ... i.e., that magnetometer and accelerometer are close to each other ...

Any thoughts?

Thanks,
Alex.

digitalentity, 2015-10-13:
This is a support question. Please use IRC or RCGroups for support to avoid diverting developer time.
Ref your problem - its likely a magnetometer misalignment. See `align_mag` variable in CLI. Set it to proper value, re-calibrate and you'll be good to go.

Shadow-MasterTf, 2015-10-20:
Take your gps apart and have a look at the magnetometers location / orientation I bought 1 of these too and found it was mounted turned 270º and it was on the bottom of the gps (upside down) I set align_mag = 8 in CLI did the trick for me.

Default orientation of X forward, Y right, Z up is 1. After that, the numbers are as follows, and mean sensor is rotated. CW0_DEG=1, CW90_DEG=2, CW180_DEG=3, CW270_DEG=4, CW0_DEG_FLIP=5, CW90_DEG_FLIP=6, CW180_DEG_FLIP=7, CW270_DEG_FLIP=8. This should cover all orientations. When carefully understood, these values can also be used to rotate (in 90deg steps) or flip the board.

Shadow-MasterTf, 2015-10-20:
Oh and like digitalentity said make sure you re-calibrate after changing the settings

closed 2017-02-25 by hydra