cleanflight/cleanflight #1387: typedef struct

ledvinap, 2015-10-09:
Using structs and typedefs is a bit problematic now.
In most code 
```
typedef struct some_s {
 int data;
} some_t;
```
is declared in `.h` file. This works fine as long header file is always included before use.

But in some cases forward declaration of `some_t` is desirable - structure may be opaque for code that uses it if it is always handled by pointer.

Currently most headers either 
- forward declare struct and use `struct some_s *`. Function declaration and definition are different. 
```
// in .h
struct some_s;
void same_fun(struct some_s* param);

// in .c
void same_fun(some_t* param);
```
- require (unnecessary) inclusion of header file with full struct declaration (not including header files directly makes this even more annoying). This approach is impossible in some cases (forward declaration can't be avoided).

IMO there are two possible improvements:
- avoid typedef, use `struct some *` or `struct some_s *` everywhere. https://www.kernel.org/doc/Documentation/CodingStyle chapter 5.
But this will mean lot of code changes and CF is generally diverging from this style.

- use typedef for forward declarations:
 ```
// in .h
typedef struct some_s some_t;
void same_fun(some_t* param);
```
 Problem with this approach is that `-pedantic` will emit warning if some_t is redefined. IMO only way around this is to drop `-pedantic`.
It may be possbile to use `#define / #ifdef` to avoid redefinition (or move typedef into separate header file and use `#pragma once`), but that will require lot's of useless code.

Does anyone have some other ideas? 

@hydra: What do you think? Is it time so choose one approach?

MauroMombelli, 2015-10-11:
i'll vote for removing typedef

tracernz, 2015-10-11:
Typedefs should be reserved for when you really mean to declare a type anyhow. I tend to agree with Linux in Chapter 5: https://www.kernel.org/doc/Documentation/CodingStyle.

ledvinap, 2015-10-21:
@hydra: Any opinion on this?

closed 2017-02-25 by hydra