cleanflight/cleanflight #1388: Processing Acc data not necessary in acro mode before motor write

borisbstyle, 2015-10-09:
@ledvinap @hydra 
I am thinking to add simple code that skippes ACC processing when in acro mode. depending weather flightmodes activated the acc reading will happen delayed so the level modes can be activated at all times.
This reduces delay before sensor reading to motor write with pretty significant numbers as acc reading and processing can take quite a while with i2c

Here is an example I used for testing
https://github.com/borisbstyle/cleanflight/blob/betaflight/src/main/mw.c#L767-L771

Any suggestions or remarks about this implementation to add it to CF before I make a PR?

ledvinap, 2015-10-09:
Even modes that take ACC into account may benefit from delaying ACC and MAG processing. Changes in attitude are based on gyro reading, ACC and MAG are only used to avoid drift and are heavily filtered. Delay caused by postponing processing after mixer will be totally negligible.

BTW: Is there really any improvement with such short loop times?

borisbstyle, 2015-10-09:
You may be right indeed. There is already a PR somewhere with faster ACC processing as well.

This was indeed a simple/safe approach to fit it in the current CF looped style. 

Loop time of 1000us allows full 1khz processing to prevent aliasing etc. I dont think the motor write is really relevant in there, but easy to prevent aliasing on that side too.
It helps reducing delay. In my code on acro mode with overclocked i2c bus to 800khz there is just 120us delay between gyro read/process and motor write. We now have a full sync when gyro sync enabled. With acc processing added additionally to it that difference becomes much bigger.

I also added a 7 taps FIR filter (check filter.c in the branch). That really made a huge, well lets say enourmous improvement cleaning up gyro signal with pretty much minimum delay. 20db cut to frequencies above 200hz. This allowed me finally to successfully remove the 3 point averaging for D as well.
Still trying to figure out how to fit all this in CF. So I am doing it in pieces now. I am yet not sure if this is a neat approach. I would rather like to see more structured change in the code. We dont have to stick to this multiwii/baseflight legacy anymore.

closed 2016-01-04 by borisbstyle