cleanflight/cleanflight #1389: I2C Overclocking Feature Request

borisbstyle, 2015-10-09:
To improve i2c speed we could have i2c overclocking feature enabled from cli. I found that F1 targets like naze32 safely can run on 800khz mode and F3 targets seem very safe on 1000khz. This reduces latency in sensor readings.

joshuabardwell, 2015-10-09:
Given how confused people get over something as simple as acc_hardware and emf_avoidance, allowing i2c overclocking via the CLI has the potential to really mess people up. I sincerely would like to see this as an option, I just know there are going to be a zillion people on RCGroups saying, "I overclocked for awesome speed and now my copter doesn't fly."

orgua, 2015-10-09:
then call it i2c_clockfrequency and it's not so obvious. There should be a note included for tuning this value, something like: "make sure all your sensors support this clock, in doubt consult the related datasheets". It would be even nice to underclock for signalwires >1m or other troubles

tracernz, 2015-10-09:
@orgua The Invensense MPU-xxxx sensors and the HMC5883L don't support anything higher than 400 kHz so there's little point in adding it if you're going to say "make sure all your sensors support this clock, in doubt consult the related datasheets." I'm sure many of the other sensors on different targets have the same 400 kHz limit as this is the standard speed for (open-drain) high speed I2C.

The problem with doing this is that the signal rise times just aren't fast enough and the high logic level becomes dangerously close to Vih and thus there is no or negative noise margin. On F1 this cannot be solved and becomes especially bad on e.g. a full Naze where the bus capacitance is much higher. On F3 it can drive the line high using a current source driver, rather than the standard open-drain approach, giving it suitably fast rise times on SCL. The problem there is that the Invensense parts still only do open-drain when driving SDA though.

A much better option if faster speed is desired is to use hardware with the gyro on SPI. You can read the accelerometer and gyro data registers on Invensense parts at 20 MHz via SPI.

orgua, 2015-10-09:
yes, that's my point. the people will think twice before trying to tune this value. 
second part: only because higher values are not supported doesn't mean that it won't work. in my experience it's not the signal quality that limits the speedboost but the reaction-timing for the shared data-line and the i2c-response-design. I mean do the math. Signal rise times around 1µs for these 400 kHz are no problem for the F1 (or every other chip younger than 15 years) even with "load". With 10k PU on the i2c-lines my scope gave me <50ns rise- and 5ns fall-time for the naze32. Lowering the 10k PU to 1k speeds up the risetime even more. 
But yes! SPI should be standard - then this talk wouldn't exist.

tracernz, 2015-10-09:
@orgua Here are some captures from a Naze32 acro (1k pullups) at 400 kHz:
[image]
And 1200 kHz (as originally proposed for betaflight):
[image]
Unfortunately I cropped the timing information off here but it is clear that SCL struggles to reach Vih. This is an acro board without baro and mag on the bus.

orgua, 2015-10-09:
next time use a real scope, please. your logic pro is fine for slower signals but changes the signal a lot because of the filtering and the long wires! do you need proof?

orgua, 2015-10-09:
400 kHz with 10k PU (the probe is interfering with the signal too, but not like the logic pro):

[image]

tracernz, 2015-10-09:
Unfortunately I don't have any room to keep my oscilloscope (and most of my other equipment) around while at university so I have to make do with the Logic at home. I'm not entirely convinced the Logic probes make that much difference to the signal amplitude but I can't say for sure. I'll go to the university later if I get a chance and measure with a real scope. I'm curious where you got a naze with 10 k pullups? Revs 2, 3 & 4 all show 1 k on the schematic and the rev 5 one in front of me is also 1k (marked 01B => 1 k 1%).

orgua, 2015-10-09:
it's a naze32rev5. and yeah, 10k are and were build in. Every distributor has it's own kind of cloning. It's not that odd. Especially when i tell you that there were capacitors mounted where the voltage-divider for measuring the batteryvoltage should have been....
But yes, you should definitely try a scope! don't get me wrong, i like saleae and their creative minds mark and joe, but this new device is not made for analogue readings (with the typical wiring). had to learn it myself.

borisbstyle, 2015-10-16:
Added for F1 and F3 targets
https://github.com/cleanflight/cleanflight/pull/1402

martinbudden, 2016-04-02:
@borisbstyle , can you close this issue?

closed 2017-02-25 by hydra