cleanflight/cleanflight #1390: SPRACINGF3 ARMING ISSUE on 1.10 CF with SBUS

thunderstrikeTT, 2015-10-11:
1) FC working on 1.9.0 CF with frsky x4R reciver with SBUS on UART3.
2) FC upgraded to 1.10.0 CF , FC manually configured and when arming FC . The FC lockup with a continuous beep tone sound and a constant red and blue light, all power needs to be cut to regain connection to the FC. This is with frsky x4R reciver with SBUS on UART3.
3) Change the frsky x4R reciver with SBUS to UART 2 and the problem still persist it keeps saying packet error at the bottom..
setting dump
Current settings: 
looptime = 2500
emf_avoidance = 0
mid_rc = 1500
min_check = 1100
max_check = 1900
rssi_channel = 7
rssi_scale = 30
rssi_ppm_invert = 0
rc_smoothing = 1
input_filtering_mode = 0
min_throttle = 1150
max_throttle = 1850
min_command = 1000
servo_center_pulse = 1500
3d_deadband_low = 1406
3d_deadband_high = 1514
3d_neutral = 1460
3d_deadband_throttle = 50
motor_pwm_rate = 400
servo_pwm_rate = 50
retarded_arm = 0
disarm_kill_switch = 1
auto_disarm_delay = 8
small_angle = 25
fixedwing_althold_dir = 1
reboot_character = 82
gps_provider = 0
gps_sbas_mode = 0
gps_auto_config = 1
gps_auto_baud = 0
gps_pos_p = 15
gps_pos_i = 0
gps_pos_d = 0
gps_posr_p = 34
gps_posr_i = 14
gps_posr_d = 53
gps_nav_p = 25
gps_nav_i = 33
gps_nav_d = 83
gps_wp_radius = 200
nav_controls_heading = 1
nav_speed_min = 100
nav_speed_max = 300
nav_slew_rate = 30
serialrx_provider = 2
spektrum_sat_bind = 0
telemetry_switch = 0
telemetry_inversion = 1
frsky_default_lattitude =  0.000
frsky_default_longitude =  0.000
frsky_coordinates_format = 0
frsky_unit = 0
frsky_vfas_precision = 0
hott_alarm_sound_interval = 5
battery_capacity = 0
vbat_scale = 110
vbat_max_cell_voltage = 43
vbat_min_cell_voltage = 33
vbat_warning_cell_voltage = 35
current_meter_scale = 400
current_meter_offset = 0
multiwii_current_meter_output = 0
current_meter_type = 1
align_gyro = 0
align_acc = 0
align_mag = 0
align_board_roll = 0
align_board_pitch = 0
align_board_yaw = 0
max_angle_inclination = 500
gyro_lpf = 42
moron_threshold = 32
gyro_cmpf_factor = 600
gyro_cmpfm_factor = 250
alt_hold_deadband = 40
alt_hold_fast_change = 1
deadband = 0
yaw_deadband = 0
throttle_correction_value = 0
throttle_correction_angle = 800
yaw_control_direction = 1
pid_at_min_throttle = 1
yaw_motor_direction = 1
yaw_jump_prevention_limit = 200
tri_unarmed_servo = 1
servo_lowpass_freq = 400
servo_lowpass_enable = 0
default_rate_profile = 0
rc_rate = 90
rc_expo = 65
rc_yaw_expo = 0
thr_mid = 50
thr_expo = 0
roll_rate = 0
pitch_rate = 0
yaw_rate = 0
tpa_rate = 0
tpa_breakpoint = 1500
failsafe_delay = 10
failsafe_off_delay = 200
failsafe_throttle = 989
failsafe_kill_switch = 0
failsafe_throttle_low_delay = 100
rx_min_usec = 885
rx_max_usec = 2115
gimbal_mode = 0
acc_hardware = 0
acc_lpf_factor = 4
accxy_deadband = 40
accz_deadband = 40
accz_lpf_cutoff =  5.000
acc_unarmedcal = 1
acc_trim_pitch = 0
acc_trim_roll = 0
baro_tab_size = 21
baro_noise_lpf =  0.600
baro_cf_vel =  0.985
baro_cf_alt =  0.965
baro_hardware = 0
mag_hardware = 0
mag_declination = 0
pid_controller = 2
p_pitch = 40
i_pitch = 30
d_pitch = 23
p_roll = 40
i_roll = 30
d_roll = 23
p_yaw = 85
i_yaw = 45
d_yaw = 0
p_pitchf =  1.500
i_pitchf =  0.400
d_pitchf =  0.030
p_rollf =  1.500
i_rollf =  0.400
d_rollf =  0.030
p_yawf =  2.500
i_yawf =  1.000
d_yawf =  0.000
level_horizon =  3.000
level_angle =  5.000
sensitivity_horizon = 75
p_alt = 50
i_alt = 0
d_alt = 0
p_level = 90
i_level = 10
d_level = 100
p_vel = 120
i_vel = 45
d_vel = 1
yaw_p_limit = 500
dterm_cut_hz = 0
pterm_cut_hz = 0
gyro_cut_hz = 0
blackbox_rate_num = 1
blackbox_rate_denom = 1
blackbox_device = 1
magzero_x = 0
magzero_y = 0
magzero_z = 0

fakatwork, 2015-10-11:
Can't reproduce using FrSky TFR8-SB on UART3+S.BUS and non-deluxe board.

Did you update your X4R firmware?

74656b, 2015-10-11:
Have a deluxe board & X8R on UART3 w/ S.BUS, no issues in 1.10 upgrade.

thunderstrikeTT, 2015-10-11:
No I didn't

Sent by Outlook for Android

https://github.com/cleanflight/cleanflight/issues/1390#issuecomment-147230318

thunderstrikeTT, 2015-10-12:
finally got it to work I had to kill all the led, telemetry and blackbox info. When I did that the FC worked properly on UART 3.Next Item was my OSD .. micro minim OSD doesn't work well with this FC it keeps restarting or freezing  ( running the mwosd1.5). The struggle continues

thunderstrikeTT, 2015-10-12:
Trying now to get Telem to work on UART 2 but the Tranis only seeing the RSSI no voltage...

borisbstyle, 2015-10-12:
I use X4RSB and Smartport telemetry on UART2 without issues on deluxe and acro SpracingF3 board

thunderstrikeTT, 2015-10-12:
Whats steps did you use I followed painless 360 and not getting any thing other than. Rssi.. I'm running a 4s lipo

borisbstyle, 2015-10-15:
I think you need to configure your Radio properly.

closed 2017-02-25 by hydra