cleanflight/cleanflight #1392: In TriCopter configuration, Servo Values Not Saved on Naze FC

mrduncan626, 2015-10-13:
In TriCopter configuration, Yaw Servo Direction, MIN, MID, and MAX are not being saved.  This problem also existed in v1.9.  Still having to configuration TriCopter in V1.8.1.

stronnag, 2015-10-13:
You can use the CLI, where configuration works for tri on current (and previous) versions.

mrduncan626, 2015-10-13:
Ok, is the following correct:
Yaw Servo MID: servo_center_pulse = 1500.
Servo direction:  yaw_direction = 0  or yaw_control_direction = 1.
It's not obvious what parameters to change for Yaw Servo MIN and MAX.

stronnag, 2015-10-13:
To change the servo direction, for 1.9, 
```
servo 5 1020 2000 1500 101 -1
````
The 101 sets reversed, for some 1.9 iterations there are some extra parameters, but the 100 / 101 defines servo direction.
````
servo 5 1000 2000 1500 90 90 101 -1
````
For 1.10 / 1.11 use the smix command set (as Mixed.md), thusly:
````
# mixer
mixer TRI
mmix reset
smix reset

# then per profile
smix reverse 5 2 r
````
min /max are probably the 1020 / 2000 values, but you'd have to check the code, as servos have not been well supported in the UI as FC code improvements have somewhat out-paced UI developer time.

mrduncan626, 2015-10-13:
Wow.....setting up tricopter servos used to be easily done directly through Servo Tab in the CleanFlight GUI.  Thanks for the information.  I'll have to study up on the Mixer.md.

mrduncan626, 2015-10-13:
It also looks like any configuration which requires servos will require the method above to setup the servos.

stronnag, 2015-10-14:
smix is much more flexible than the older methods, just needs the UI to catch up.

closed 2017-02-25 by hydra