cleanflight/cleanflight #1393: PID Controller 2 Luxfloat: ROLL & PITCH rates don't work in Angle

skaman82, 2015-10-14:
I have played a bit with Luxfloat. So far I like it very much. But in angle/horizon mode the quad reacts very slow. I tried to play around with ROLL & PITCH rates. In horizon this has the desired effect and the quad reacts now how I like it. But in ANGLE mode it still behaves very poorly like it did with ROLL & PITCH rates where down to zero. Is this a bug or I'm doing something wrong?

borisbstyle, 2015-10-14:
@hydra 
Yes thats a known issue. There is a limitation of max angle declanation to the rc command.
See here: https://github.com/cleanflight/cleanflight/issues/1072

Someone suggested a fix earlier
https://github.com/cleanflight/cleanflight/pull/1180

But there is a better one here
See here.
https://github.com/borisbstyle/cleanflight/blob/betaflight/src/main/flight/pid.c#L146-L150

Current Code:
https://github.com/borisbstyle/cleanflight/blob/betaflight/src/main/flight/pid.c#L146-L150

skaman82, 2015-10-14:
Thanks. Now as it is the ANGLE mode is practically not flyable with luxfloat. 
I think this should be marked critical.

borisbstyle, 2015-10-14:
@skaman82 This has always been like this actually. It is totally flyable, but you have to increase your rccommand to maximal value to get most response, which is still not that great, but flyable.

skaman82, 2015-10-14:
ok, but it would be also great when someone could fix this :)

borisbstyle, 2015-10-14:
@ctzsnooze was working on that PR I believe. Not sure what the current status is

borisbstyle, 2015-10-14:
Can you in the meanwhile try this test code and verify if this behaviour is right
https://dl.dropboxusercontent.com/u/31537757/cleanflight/betaflight/betaflight_NAZE.hex

skaman82, 2015-10-14:
I will test you HEX file tomorrow and will get back to you.

borisbstyle, 2015-10-14:
so your full stick is always max angle configured which is 55degrees. For more angle just configure it in the cli.

skaman82, 2015-10-14:
Had a little time and tested your HEX right away. I am not seeng any difference in described behavior. ANGLE still not accepting ROLL & PITCH rates. In HORIZON it works fine. With ROLL & PITCH rate value of 0.20 I have a good crisp reaction - even to smaller stick inputs. In ANGEL it reacts very differently and smushy. The Stick sensitivity is complete different and isn't changeable by increasing ROLL & PITCH rates. Raising RC Rate to 2.0 and setting ROLL & PITCH rates to 0 helped achieve desired behavior but I think this is not how it should be done. My goal is that I can change sensitivity to ROLL & PITCH independently and the quad reacts to it the same way in ANGLE lite it does in HORIZON – only with angle limitation. It works very well on PID Controller 3.

borisbstyle, 2015-10-14:
@skaman82 rates have nothing to do with it
set max_angle_inclination = 500 means max stick 50 degrees

set max_angle_inclination = 800 means 80 degrees

So when you give full stick command it will match that angle.

skaman82, 2015-10-14:
max_angle_inclination variable should have nothing to do with stick sensitivity. I usually set max_angle_inclination to 350 on my smaller quads to reduce speed without loosing stick sensitivity. I honestly don't know why this is handled differently than for example on PC3 where max_angle_inclination just defines maximal angle and ROLL & PITCH rates define sensitivity. ROLL & PITCH rates should behave the same way through all flight modes.

borisbstyle, 2015-10-14:
It is meant to use the full stick. On your method you are only using 35% of your entire stick movement. So there is dead stick there.

To increase sensitivity you could add expo.

Horizon mode is half angle and half acro and that is different. There is no point of throwing away stick range when you are not using it.

skaman82, 2015-10-14:
I think that is exactly my point. When using max_angle_inclination=350 it should just limit the maximum angle and not the stick input. You are saying "Horizon mode is half angle and half acro and that is different." But why my desired ROLL & PITCH rate configurations work on other multiwii controllers through all flight modes but not in luxfloat? Is there a reason? So as it is is looks broken or not finished to me as a normal user. Or at least not usable like I expect. 

Also it would be nice if the PID Controllers had a consistent way to setup in the GUI or at least a adapted interface to change parameters depending on the selected controller with right labeling what each value does. It is a living nightmare for normal users how it is. But that is another discussion.

borisbstyle, 2015-10-14:
It sounds like you need to use Horizon mode :dancer: 

luxfloat is actually a first non-multiwii pid controller and uses the same method like any other FC on this part. Like taulabs, openpilot, DJI etc.
There is no point of having the same behaviour on all pid controllers as there wouldnt be any advantage of having them all.
If you like multiwii standard behaviour you should use PID3

skaman82, 2015-10-14:
I think the best way to do this is measure the angle during flight and ignore stick inputs that would push the quad over the defined max_angle_inclination value. For example: max_angle_inclination=500. When the quad reaches the 50° at lets say 60% stick input and it receives 100% input, it ignores the 40%. This way we don't loos stick sensitivity by reducing max_angle_inclination. Meanwhile I stick to PID3.

borisbstyle, 2015-10-14:
@skaman82 
Thats exactly why there are many pid controllers to fit everyone's choices :dancer: 

But honestly horizon mode covers all that. There is no max angle limit and you can get max sensitivity. You can even do flips or rolls if you want.

borisbstyle, 2015-10-14:
in luxfloat it's the purpose to get the maximum resolution/precision for all modes

skaman82, 2015-10-14:
I usually fly in ANGLE an switch to HORIZON for flipping. I personally like the max angle limit on full stick.

borisbstyle, 2015-10-14:
Yeah but even your half stick is same like full stick  :D.

skaman82, 2015-10-14:
Thats why I compensate with ROLL & PITCH rates :D

ctzsnooze, 2015-10-15:
@skaman82

I agree entirely with you, that stick sensitivity should not be linked to max angle.

The following code fixes the max angle bug and keeps stick sensitivity independent of max angle:

https://github.com/ctzsnooze/cleanflight/commit/6850b766fd8bc9cf82299d1f4d55e6231eafb7e2

Boris's alternative solution (currently implemented in betaflight) is that max_angle is always attained at maximum stick deflection.  This results in very weak stick responses at centre when max_angle is low.  

In contrast, my solution keeps stick responsiveness completely separate from max_angle.  In practice my solution works quite well, if you have rates that are OK for acro or horizon modes, level mode will feel OK around centre sticks, regardless of max angle.

Neither alternative has been implemented in Cleanflight to date, maybe because no-one agrees on the best solution yet.

borisbstyle, 2015-10-15:
@ctzsnooze He uses pid controller 3 now which offers solution for his problem :D. That's a good thing of having multiple pid controllers to fit everyone it's taste.

skaman82, 2015-10-15:
@borisbstyle Yes it is a good thing to have multiple controllers but I think the have to be consistent ways to setup them. It just a bad design when one variable does different things on different controllers. In that case there has to be a different variable (in the GUI a correct labeled field ore none if not used). Still don't get boris why the R/P rates configuration is intentionally limited for ANGLE on luxfloat compared to other pc. Makes no sense to me personally. But this is my sight of things...

borisbstyle, 2015-10-15:
@skaman82 When they all would fly the same you might as well remove them and keep only 1 pid controller.

borisbstyle, 2015-10-15:
@skaman82 Its only the folks who are coming from the multiwii world who think level would need to be like that, while every other FC has the same approach where full stick resolution is being used.

skaman82, 2015-10-15:
@borisbstyle I think you misunderstand. I don't mean the math behind the different controllers. I just mean the design of the configuration variables. If max_angle_inclination is used to limit the angle in other PCs without affecting the rates the lux controller should use another variable instead. Also R/P rates shouldn't  affect HORIZON and ANGLE in your approach. It would be a logical thing. Bu they do.

digitalentity, 2015-10-15:
I personally think that ANGLE mode should reach max_angle_inclination at full stick and roll/pitch rates shouldn't affect it at all.
HORIZON, however is different as it is a combination of ACRO and ANGLE and is very implementation-dependent. I prefer it to be ACRO + weak leveling without max_angle_inclination imposed limit.

skaman82, 2015-10-15:
I personally don't mind the approach. But I personally don't like the fact that when we use low max_angle_inclination value with R/P rates the quad flies fine and twitchy in HORIZON but moves like a sand bag in ANGLE.

ctzsnooze, 2015-10-15:
It was a simple bug in luxfloat. I do not think it was intended the way the code was written.

Yes all multiwii code works as skaman82 describes.  

Having tried both solutions and trained people on both, I disagree with digitalentity.  It is nearly impossible to fly level mode in luxfloat with max_angle at default values with stock cleanflight code, because it is just ridiculously stick insensitive.  The transition from level to Horizon is too big.  Having taught lots of beginners with normal stick sensitivity but limited angle it is very easy.  They learn for proper control not to make big stick movements, and they don't get into trouble.  Requiring them to make huge stick movements to get a little movement of the quad makes the transition to horizon or acro much harder.  

For me I don't care mostly my acc are disabled any way.

ctzsnooze, 2015-10-15:
Like a sand bag is very good description.  Un-flyable I say.  Even - especially - for beginners.

skaman82, 2015-10-15:
@ctzsnooze "The transition from level to Horizon is too big." Thats exactly my problem with it.

borisbstyle, 2015-10-15:
@ctzsnooze The early success of openpilot was exactly this thing that everybody liked the logical angle mode! I remember in my beginner times that I exactly like the predictable stick sensitivity where full stick was simply the max configured angle. I never could get used to the multiwii weird angle approach as many others I knew back than.
Its not for nothing that every other FC software does the same.

borisbstyle, 2015-10-15:
@skaman82 The current luxfloat approach is not even both of the mentioned btw :D

skaman82, 2015-10-15:
The only way I see to fly with max angle of 35° and make the transition between ANGLE and HORIZON more fluid is not to to use R/P rates at all and crank up the RC rate to an immense value. It works ok but it is not logical from user point of view to set the RC rate to 2 to make it work!

borisbstyle, 2015-10-15:
@skaman82 In my approach rccommand has nothing to do with your sensitivity. Its only your inclination paramater. Current luxfloat approach is indeed with rccommand, which is bad IMO.

ctzsnooze, 2015-10-15:
Anyway maybe the only real solution to this is yet another CLI option... full_stick_level_mode !  :-)

skaman82, 2015-10-15:
It seems the only way to have max angle of 35° and stick sensitivity in the middle. Otherwise it is not flyable in ANGLE with max angle value of 35°. The transition between ANGLE and HORIZON should be fluid on controls independent of the max angle setting.

borisbstyle, 2015-10-15:
@skaman82 I dont think thats possible. Well it is, but horizon mode has configurable transition point for  level going into acro on luxfloat. You can configure the horizon sensitivity, which translates into where your leveling goes into acro. You cant link these 2 flight modes

skaman82, 2015-10-15:
@borisbstyle  Well that I meant when I mentioned the lack of configurable settings for ANGLE on luxfloat. On Multiwii PCs you can setup max angle in the CLI and the sensitivity how each axis reacts to stick inputs by R/P rates. All values can be configured independently. That is simply not possible with luxfloat. There just one variable for all three the above settings. Seems like luxfloat was not (fully) implemented for ANGLE flight mode but nobody seems to care.

ctzsnooze, 2015-10-16:
skaman send me a pm via rcgroups I can build you a hex with my code then you can see if it does what you want.  You may find not exactly the same sensitivity in level mode as horizon but it will be very close, and because you can change sensitivity for horizon, you can match them if you like.  My code also has a modification for Horizon that allows you to experiment with inverted flight in horizon.

phantom-8, 2016-02-03:
Today is my first time trying Luxfloat.  Acro mode is very nice, but Angle mode is unflyable.  The stick is very sluggish and hard to control.  The symptom is exactly as skaman82 described.  I didn't change max_angle_inclination and left it as default.  I also use Openpilot CC3D and never encountered this problem.

closed 2017-02-25 by hydra