cleanflight/cleanflight #1394: Spurious beeps and glitches on throttle channel using Spektrum satellite Rx

AndersHoglund, 2015-10-14:
I have had spurious beeps and motors stopping when using Spektrum serial Rx satellites. Also seen as glitches in the Rx tab of the configurator.

Started seening this with early october builds of Boris betaflight 1.10. Later also seen with Cleanflight 1.10 release. Cleanflight 1.9 seems OK. Betaflight 1.11-final still have this problem.

I have seen this on two different rigs using Naze32 acro and Flip32 acro with three different satellites, OrangeRx DSM2, LemonRx DSM2 and Spektrum DSM2. Using SPEKTRUM_1024 (bind value 3) seems worse, using SPEKTRUM_2048 (5) seems better. Not 100% OK though. Could be coincidence, too few tests to tell for sure.

Now also running LemonRx DSMx sateliites with long antennas on two rigs using DSMx 11ms (9), no glitches seen so far. But I have only had a couple of flights yet, too early to tell. 

See discussion and screenshots in the betaflight RCG thread, starting here: http://www.rcgroups.com/forums/showthread.php?t=2464844&page=353&highlight=Orx#post32896393
Sample screenshot of Cleanflight 1.10 also here: https://dl.dropboxusercontent.com/u/11080368/cleanflight_151012_hickup_DTI_Blade180.jpg
All screenshots taken zipped up here: https://dl.dropboxusercontent.com/u/11080368/cleanfligh_throttle_glitch_screenshots.zip

Could this be related to the Sbus drops outs also under discussion? Same serial rx interface.
/A

borisbstyle, 2015-10-14:
@ProDrone 
Can you build a debug version for him to determine whats causing it?

borisbstyle, 2015-10-14:
@AndersHoglund 
Can you in the meanwhile test the suggested settings from prodrone to see if you can get 1.9 behaviour
rx fail and failsafe are highly configurable and for race quads i would recommend to set

`rx fail` for stick channels 0 - 3 to hold,
`failsafe_delay` to 5,
`failsafe_off_delay` to 0,
`failsafe_throttle` to 1000.
That way you should not notice single signal drops (cause that is what it looks like) and after 700 ms (200ms fixed + 500ms delay) failsafe handling will activate and drop the quad followed by immediate disarm. That is what most fast-lane users prefer i think.

ProDrone, 2015-10-14:
> Can you build a debug version for him to determine whats causing it?

@borisbstyle 
Which source for what target do you mean/need?

borisbstyle, 2015-10-14:
@ProDrone What you think it's best to debug this :D.

AndersHoglund, 2015-10-14:
Tried the suggested "Fast Lane"-settings above and did not see nor hear a single glitch in several minutes. Then switched the Tx off to se failsafe behaviour, and yes this is what I like. Screenshot at: https://dl.dropboxusercontent.com/u/11080368/betaflight_151014_Failsafe_behaviour_fast_lane%20settings_DTI_Blade180.jpg

AndersHoglund, 2015-10-14:
I have a feeling things have improved in betaflight 1.11 final, so to check that my rig still has the problem I re-ran with old values delay=10, off_delay=200. And I only saw a couple off very small throttle glitches and no beeps. Will revert to CF1.10 and re-check again.

Q: Are we chasing a failsafe or a serial data handling problem? Maybe both...

AndersHoglund, 2015-10-14:
Oh yes, 1.10 stable is much worse. Or it is all coincidental... 
I cought four glitches in the same screenshot, one very very small: https://dl.dropboxusercontent.com/u/11080368/cleanflight_151014_hickup_DTI_Blade180.jpg

joshuabardwell, 2015-10-14:
Is that with default rxfail settings or "fast lane"?

borisbstyle, 2015-10-14:
@AndersHoglund 
I will update betaflight in few minutes with reduced rxfail sensitivity. Hang on :D

borisbstyle, 2015-10-14:
While my script is busy with compiling here the latest hex for NAZE target. Please test this one
https://dl.dropboxusercontent.com/u/31537757/cleanflight/betaflight/betaflight_NAZE.hex

AndersHoglund, 2015-10-14:
Took a while but sorry to say, there it was again: https://dl.dropboxusercontent.com/u/11080368/betaflight_151014_hickup_DTI_Blade180_BorisB-Failsafetest2.jpg
Difference from final beeing? Failsafe_off_delay default value??

ProDrone, 2015-10-14:
@AndersHoglund So it has improved but is not gone.
@borisbstyle In search of the right value i think `#define MAX_INVALID_PULS_COUNTS` can be set to 6 instead of 3 (and then make a finalfinalfinal version of betaflight (you should really invent some naming convention for that :)

AndersHoglund, 2015-10-14:
Running prodrones testbuild as we speak, looking good so far... 5minutes no glitch. On default failsafe values.

borisbstyle, 2015-10-14:
@ProDrone The version only relates to my flight behaviour changes. All other CF changes are not related to version. It doesnt change the behaviour.

AndersHoglund, 2015-10-14:
Ahh, sorry guys. I spoke too soon. Got a glitch in prodroness testversion as well; https://dl.dropboxusercontent.com/u/11080368/betaflight_151014_hickup_DTI_Blade180_prodrone-Failsafetest1.jpg

borisbstyle, 2015-10-14:
@AndersHoglund 
Can you try this one
https://dl.dropboxusercontent.com/u/31537757/cleanflight/betaflight/betaflight_NAZE.hex

AndersHoglund, 2015-10-14:
Hope I'm not opening my mouth to soon again, but that last version is looking good.
14 minutes, no glitch. I'll buy this one. But the problem with spourious errors are when to say OK? If we had failures by some minutes, will 10 timmes or 100times that do for passing test verdict? Classic problem.

 Will take it for some outdoor IRL tests tomorrow, weather permiting. Thanks guys, great support, and a fun session. Signing of, nighty nighy

borisbstyle, 2015-10-14:
@ProDrone @AndersHoglund 
I know used value of 20 in the above hex. Very curious to how it will go. I think we can use such values in here as that's not that much time at all. This is now in my last betaflight binaries.

I would also be curious to get the counts visible in debug. Will build a version for you tomorrow so wel can collect more data about it, but I have a feeling we are good now.

ProDrone, 2015-10-14:
@borisbstyle Wow, did you jump from 3 to 20? It is a good mod if it keep those quads in the air but i still want to know what is going on in there...

borisbstyle, 2015-10-14:
Well calculate how much time that is. That is not that much at all.

joshuabardwell, 2015-10-15:
@borisbstyle is your most recent link above flight-ready? I will give it a try also if so.

joshuabardwell, 2015-10-15:
I have monitored about 20-30 minutes in the Receiver tab and the throttle channel has been rock solid. Incidentally, this is with the default rxfail configuration, so not even changing to rxfail 3 h. 

I guess I will give a try at flying it tomorrow! Fingers crossed.

borisbstyle, 2015-10-15:
@joshuabardwell Yes that's the latest βf

AndersHoglund, 2015-10-15:
Four rigs in different configs now flown on this latest greatests Boris "failesafetest2" version. No problems seen. Total air time only about one hour though, due to other tasks and issues.
Q: Is this the f399e455ebbfddf9af79314c642912c6653d6aab-commit on the betaflight_dev branch?

joshuabardwell, 2015-10-15:
Three batteries just now, about 12 minutes total flight time, with one battery spent just flying gently for about six minutes, and two batteries spent thrashing a bit harder. No dropouts. So far, so good.

As usual, immense thanks to @borisbstyle for his work on this issue. Really glad to be able to fly on my X4R finally.

ProDrone, 2015-12-03:
Should be solved with
https://github.com/cleanflight/cleanflight/issues/1394
https://github.com/cleanflight/cleanflight/pull/1374
and several other pull requests since then.

closed 2015-12-03 by ProDrone