cleanflight/cleanflight #1395: CF 1.10.0  RX Dropouts / Glitches not just on frsky sbus

Tabsy123, 2015-10-14:
Some info if needed

It seems the problem may be more wide spread than just being with sbus

http://www.rcgroups.com/forums/showpost.php?p=32940118&postcount=5897

note:  both people in that thread had cf failsafe unticked, if that may help with diagnosis on 1.10.0 issues etc

fakatwork, 2015-10-15:
I had a nasty brownout yesterday on my FrSky TFR8-SB and I thought it was due to me flying too close to high tension lines, but it could have been the problem in retrospect... The 1.10 release notes are very elusive. It's not clear whether it's just some FrSky S.Bus models or all of them.

I just know for sure that I had zero issue with same setup on 1.9. I fail to understand how a "frsky s.bus bug" could appear out of the blue, unless 1.10 unmasked it somehow and it was present before? Then is it really a FrSky bug and not a CF 1.10 bug? Depends on point of view I guess...

So to move on, and I'm just throwing ideas here so we can advance, could it be linked to the recently introduced rc_smoothing? I had it enabled, what about you guys? Let's try to see if there is any correlation.

borisbstyle, 2015-10-15:
This has been resolved in 1.11
https://github.com/cleanflight/cleanflight/issues/1394
https://github.com/cleanflight/cleanflight/pull/1374

You can try these binaries in the meanwhile:
https://dl.dropboxusercontent.com/u/31537757/cleanflight/betaflight/betaflight.zip

Tabsy123, 2015-10-15:
Thanks Boris where can i find / download 1.11  for naze?

borisbstyle, 2015-10-15:
Zipfile in the link has all betaflight tested hex files. It's a bit different than cleanflight, but based on 1.11 build.

Tabsy123, 2015-10-15:
Thanks heaps mate

ProDrone, 2015-12-03:
Should be solved as mentioned above.

closed 2015-12-03 by ProDrone