cleanflight/cleanflight #1396: Feature Request: CLI partial_dump

powdermnky007, 2015-10-15:
There should be a partial dump command that doesn't output everything.  This would prevent users from blindly overwriting updated defaults when updating to a new firmware version.  It should spit out any commands that have been changed from default values.  Something like this:  

mixer QUADX
mmix reset
smix reset

feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature RX_PPM
feature VBAT
feature MOTOR_STOP
feature FAILSAFE
feature TELEMETRY
feature ONESHOT125

set dterm_cut_hz = 35

set pid_controller = 2
set p_rollf =  1.300
set i_rollf =  0.320
set d_rollf =  0.015
set p_pitchf =  1.500
set i_pitchf =  0.350
set d_pitchf =  0.026
set p_yawf =  3.500
set i_yawf =  0.900
set d_yawf =  0.008

set rc_rate = 100
set rc_expo = 70
set rc_yaw_expo = 40
set thr_mid = 50
set thr_expo = 0
set roll_rate = 35
set pitch_rate = 35
set yaw_rate = 35
set tpa_rate = 55
set tpa_breakpoint = 1450
set pid_at_min_throttle = 1

map TAER1234
set align_board_yaw = 90
serial 0 5 115200 57600 0 115200
serial 1 0 115200 57600 0 115200

aux 0 0 0 900 1200
aux 1 12 0 1600 2100

skaman82, 2015-10-15:
Like this idea. This should also include:

deadband
yaw_deadband
align_mag
max_angle_inclination 
blackbox_device
blackbox_rate_num
blackbox_rate_denom

joshuabardwell, 2015-10-15:
CF knows what the default values for each command are. It should be possible to output only the things that have changed from default, rather than pre-specifying which things are and are not included.

ledvinap, 2015-10-15:
@joshuabardwell : It's not that simple - default values are represented in code as assignments, so simple comparison is impossible ...

skaman82, 2015-10-15:
even better

powdermnky007, 2015-10-15:
That would be the most graceful way to go about it IMO.

powdermnky007, 2015-10-15:
To keep from adding another cli command, you could give the dump command some parameters. 
Dump                      = keep old behaviour
Dump common     = dump common options
Etc...

skaman82, 2015-10-15:
IMO It would actually be best if the dump command would be not necessary at all and all settings would be saved via the backup button in the GUI. Thats why the Button is there in the first place, right? Later we could add a second "backup common options" button.

andymac4182, 2015-10-15:
It would be good to get this added to the issue around redesigning the protocols for talking to the FC. https://github.com/cleanflight/cleanflight/issues/347

closed 2017-02-25 by hydra