cleanflight/cleanflight #1398: RX - Make SBUS inversion configurable from CLI

ProDrone, 2015-10-16:
This will make it easy to use (inversion) modified receivers on F1 and F3 targets.

andymac4182, 2015-10-17:
Is it possible to just auto detect this?

ProDrone, 2015-10-17:
Maybe, it is what i am aiming for... but i expect it will be using a CLI setting (at least at first).

borisbstyle, 2015-10-17:
@ProDrone 
Do you mean to be able to disable inversion on F3 targets? I don't see this as a valuable feature actually.
Modified receiver still can work on normal SBUS. In fact you can modify them in such way that you can use both.

ProDrone, 2015-10-18:
Yes, i mean disable/enable the inversion for SBUS via a CLI setting.

It is of course possible to add a second output with the non-inverted signal, but many modify their receiver pemanently to bypass the inverter, like so http://www.rcgroups.com/forums/showpost.php?p=32966020&postcount=6080

So instead of changing it back you may simply set inversion off.

Maybe also useful for OpenLRS (did not looked into that...)
http://www.rcgroups.com/forums/showpost.php?p=32952765&postcount=5967

ProDrone, 2015-10-19:
@hydra Do you agree its useful (before i progress) ?

bdoiron74, 2015-10-19:
I see this as useful, since I removed the inverter from my SBUS Rx... @borisbstyle Yes it 'can' be modified to do both, but that doesn't mean it was ;)

borisbstyle, 2015-10-19:
Sorry I am still kind of confused. Bypassing inverter on X4RSB doesnt require "removing" of inverter.

I also have hacked X4RS receivers and I can use them for both scenarios. With and without inverter.

borisbstyle, 2015-10-19:
@ProDrone In that rcgroups link he just has to solder the white wire back to the throughole.
What I do is I solder the uninverted wire to one of the other unused channels like for example PPM. My ppm connection pins are uninverted and my normal SBUS connection is inverted.

Only other left argument would than be the OpenLRS than.

borisbstyle, 2015-10-19:
on second thought. It might be usefull to replace the current telemetry_inversion command by just inversion for individual UART's for telemetry and everything else to keep it flexible?

ProDrone, 2015-10-19:
@borisbstyle Oh, now you got me confused :confused: There are 4 outputs on X4RSB: CPPM, CH9, CH10 and SBUS, so they are all connected/assigned. I disconnect the inverter output that goes to the SBUS signal output by removing R35 (0 ohm resistor), next i route the uninverted signal to the outgoing side of R35. This is easy to restore to original by taking the wire out and shortening the R35 pads.

__BEFORE__
[image]

__AFTER__
[image]

When the proposed option is implemented you simply don't need to take your X4RSB apart for modification again. Just plug it over and disable invert.

ProDrone, 2015-10-19:
> on second thought. It might be usefull to replace the current telemetry_inversion command by just inversion for individual UART's for telemetry and everything else to keep it flexible?

Good thinking. I would like to know what @hydra thinks, maybe adding something to the serial CLI command?

borisbstyle, 2015-10-19:
@ProDrone I didnt remove any of the resistors. I have a wire going from the U8 uninverted side(middle pin) directly to the CPPM pin. I did disconnect the CPPM pin from the board. 
With this mod I have inverted connection to the original SBUS connection and uninverted to CPPM connector above SBUS connector. I basically can choose where to plug connector to.

I choose for CPPM, but you can also sacrifice CH9 or CH10 for that.

That is in my case better mod. Well for me as I have F1 and F3 boards I fly with and can easily exchange between them.

ProDrone, 2015-10-28:
@borisbstyle I think your solution to keep both (inverted & non-inverted) outputs is indeed more flexible, if you sacrifice one of the servo outputs it is even better, because then you have maximum connectivity (CPPM, SBUS and non-inverted SBUS).

hydra, 2015-11-04:
I'm not sure this is useful for the majority of cleanflight users.

Let's think about the target audience.  

1. Buy FC, buy RX, fly.  99%.
2. Buy FC, buy RX,  hack RX, tinker, tinker, tinker fly 1%.

thoughts?

hydra, 2015-11-04:
This seems like code that the tinkerers can merge into their own forks to use if they want.

Even adding auto autodetection would still be catering for the minority.

borisbstyle, 2015-11-04:
that was my thought too.

Perhaps those wrong hacks should be avoided.
A good hack is to keep them both inverted and not inverted :)

ProDrone, 2015-11-04:
@borisbstyle I have no idea how many have made a hack that needs this when they go from F1 to F3.
I agree that it can be hacked in a better way (see my comment above ;)) 

I also like simplicity and the reduction of CLI variables. I think it is better to close the PR, people who need it can now find it and merge it by themselves.

ledvinap, 2015-11-04:
what about '#define SBUS_DISABLE_INVERSION' ?
Then just use correct settings at https://github.com/cleanflight/cleanflight/blob/master/src/main/rx/sbus.c#L70.

ProDrone, 2015-11-04:
@ledvinap Do you mean to add such a define to target.h?

ledvinap, 2015-11-05:
@ProDrone : target.h is one possibility. Maybe Makefile OPTIONS variable is slightly better, user does not need to modify source code ...

ProDrone, 2015-11-08:
@hydra @ledvinap
I modified this to work as a compile-time option using `OPTIONS=DISABLE_SBUS_INVERSION`
No code will be added when the option is not used.

hydra, 2015-12-18:
Who actually needs this?

hydra, 2015-12-18:
From IRC:

```
[14:02] <ReadError> that is also a useful feature hydra_
[14:02] <ReadError> atleast in terms of some openlrs stuff
[14:02] <ReadError> only a few rx have inversion but all can output sbus
[14:12] <hydra_> ReadError: ahhh. that's different then.
[14:13] <hydra_> I'd be happy with a proper solution that allows SBUS inversion to be configured with the default to be correct for the taranis / frsky/futaba gear since that is what most SBUS users use.
```

So the long and short is, NO we don't want a compile time option, we apparently need a runtime solution.

ProDrone, 2015-12-19:
@hydra Should the inversion also be accessible via configurator?

hydra, 2016-01-08:
merged, thanks @ProDrone

closed 2016-01-08 by hydra

jcowgar, 2016-01-09:
Does this mean that a hardware inverter is no longer needed on the FC? i.e. a Naze 5 board can take SBUS input w/no changes/inline inverted? It's just a software flag now?

ledvinap, 2016-01-09:
@jcowgar : No. This is about using non-inverting (modified) SBUS receiver on F3 board ..

dkisselev, 2016-01-10:
On F3 boards every uart can be inverted in software (this is a standard hardware feature of the F3 line of st chips). This PR just adds support to CF to set whether that feature is enabled or not.

F1 boards (naze) don't have the hardware component so still need external inverters.

joshuabardwell, 2016-01-10:
Is there a separate issue / PR for supporting both inverted and non-inverted SBUS on F1 boards? I know this can be done at compile-time, so it seems like it should be doable at run-time.

ledvinap, 2016-01-10:
@joshuabardwell : 
- softserial may work (I used it before I switched to F3, both sbus and frsky telemetry), but current implementation is far too slow.
- some boards have onboard invertor, sometimes pin-configurable

but F1 UART can't work in inverted mode

joshuabardwell, 2016-01-10:
I think I see the problem. The UART itself doesn't recognize the incoming signal, and CF is interfacing with the UART digitally; CF isn't bit-banging the signal, so it can't simply invert its logic.