cleanflight/cleanflight #1400: Board documentation- STM32F3Discovery

varoudis, 2015-10-16:
Any documentation for STM32F3Discovery with pins and basic setup?
Thanks

nathantsoi, 2015-10-16:
I use the data sheet and https://github.com/cleanflight/cleanflight/blob/master/src/main/target/STM32F3DISCOVERY/target.h

varoudis, 2015-10-20:
another important questions that needs documentation is about the USB VCP driver that is needed in order to connect to a flushed stm32f3disc.

I managed to flush it but I can't get the user-usb to link.

tannewt, 2016-01-02:
I can't connect to a STM32F3Discovery either over USB. I'm on a Mac but I also tried on Linux. dmesg shows some enumeration errors but then on Mac is shows up as a tty. However, it can't load the configuration through the configurator.

tannewt, 2016-01-02:
I take this back. The VCP driver works for me on Cleanflight but not Betaflight.

nathantsoi, 2016-01-05:
stm32f3discovery isnt supported by betaflight: https://github.com/borisbstyle/betaflight/issues/95

this issue can pry be closed & moved to betaflight

closed 2017-02-25 by hydra