cleanflight/cleanflight #1403: CF V1.10 Failsafe Adnominally

mrduncan626, 2015-10-16:
Well, I see a lot of talk about failsafe issues in v1.10 but I have not come across the problem I'm having. So, I am not sure if it's a v1.10 issue and just a user problem. The failsafe worked fine in v1.8 and v1.9 though. When a test my failsafe (radio off), the motors do one of the following:

 1. Motors go to min-throttle for about a second when motor_stop is not activated and then it goes into failsafe at define failsafe_off_delay and failsafe_throttle setting.

 2. OR, motors stop or go to min_command for about a second when motor_stop is activated and then it goes into failsafe at define failsafe_off_delay and failsafe_throttle setting.

joshuabardwell, 2015-10-17:
Are you sure this is a bug and not just a changed behavior? Have you researched the rxfail command, which is specifically designed to affect the time between signal loss and failsafe activation?

mrduncan626, 2015-10-17:
Well, that was my question.....bug or not a bug.  I have 8 quads which have CF installed on and I have always exercised and verified the failsafe feature after every CF version installation. I haven't found much information on use of the rxfail command.  If it's not a bug, then what is causing the motor stoppage during a failsafe event as this does not occurred in v1.9 or previous versions. Can you provide any information on rxfail?

henn1001, 2015-10-17:
Sounds to me like it´s working as intended?

joshuabardwell, 2015-10-17:
Rxfail controls what the channels do from the moment the signal is lost to the moment when failsafe kicks in. Failsafe controls what happens after failsafe kicks in until failsafe shuts down. If your channels are not doing what you want from signal loss to failsafe, use rxfail to make them do whatever you want.

mrduncan626, 2015-10-17:
Yes, I read this in the RX doc and I can dump the rxfail parameters in CLI by just typing rxfail but I'm not sure how to change the values.  Do you use the "set" or "feature" to change the values?

joshuabardwell, 2015-10-17:
Just type e.g. "rxfail 3 h". Rxfail is its own command, not a parameter that you set.

mrduncan626, 2015-10-18:
Thanks.  "rxfail 3 h" and "rxfail 3 a" works but I get a "parse error" when I use "rxfail 3 s 1200" to preset a throttle value.  Am I using the rcfail set function correctly?

henn1001, 2015-10-18:
rxfail x s(et) works only for the aux channels.

ProDrone, 2015-12-03:
That is done for safety. Imagine setting the aux channel for your arm switch to armed AND throttle to full throttle. Then when a failsafe condition occurs your quad will take off with full throttle!

Works as intended, issue closed.

closed 2015-12-03 by ProDrone