cleanflight/cleanflight #1410: In accro/angle mode Flight performance are better if acc_hardware==1 (disable)

Sil20, 2015-10-18:
Version 1.9.0 Board Naze32 full.

The accro/Angle mode does not include Acceleromettre data  so the flight performance should not get bad whatever the parameter acc_hardware is .

I did not read the code but accro/Angle

joshuabardwell, 2015-10-18:
Angle mode cannot be enabled if acc_hardware=1. Self-level requires the accelerometer. This issue is really confusing.

Sil20, 2015-10-18:
Sorry I was confused about Angle mode. I edited the Issue. only Accro mode is concerned by the issue.

andymac4182, 2015-10-18:
With some of the latest changes from Boris this won't be required yet. This is really a question for the forums or IRC. Disabling the acc just allows lower loop times on older firmwares. If you are looking for faster loop times check out Betaflight as it includes changes that are on their way or being tested.

Sil20, 2015-10-19:
OK , Great, 
Would BetaFlight enhancement be merge into cleanfliht when validated and tested on ist too much of a change of approach?

andymac4182, 2015-10-19:
There are various Pull Requests for some of the features already that are being discussed :) There is still a lot of testing going into the latest stuff.

thenickdude, 2015-10-22:
The IMU needs to run all the time (even in acro modes) so that an initial attitude is available if you decide to switch to a self-levelling mode mid-flight.

closed 2015-10-22 by thenickdude