cleanflight/cleanflight #1412: GPS with Cleanflight on Naze32

khomouda, 2015-10-19:
I have a problem with the Naze32 10DOF, and GPS. I have 2 Neo-6M modules, that connect well to the u-centre, and get many sats, low HDOP, and a fast fix even indoors, but, when connecting to the naze, always 0 sats and no fix even after long wait outdoors.
On gpspassthrough it also does not give a fix or sats, although the green GPS button is on and all signal bars are showing signals, but unusable in baseflight, tried all baud rates. what can be the problem.

The strange thing is that I can't get debug info in baseflight and it shows that the GPS baud is 115200 although it is 57600 in cleanflight and in the GPS configuration. 

When I flashed baseflight and configured it to 57600 it started getting debug info, I returned back to cleanflight and disabled gps_auto_config before connecting the GPS, and made sure serial scenario 1 shows GPS baud rate to be 75600, when connecting to baseflight, it still reads it as 115200 and no debug info, and baseflight can't seem to be able to change it with cleanflight on-board. 

Very weird, I tried two Nazes and 2 GPS modules and the same...nb. autobaud is off.
there is no gps_baudrate or gpspassthrough_baudrdate commands in cleanflight 1.8, 1.9 or 1.10.

Uh1Pilot, 2015-11-04:
Your question is very confusing, are you running Baseflight or Cleanflight firmware on the board?

khomouda, 2015-11-04:
Cleanflight, and then I cant get proper signal, and when connecting Baseflight to it, I onlpy get unusable signals and no debug info.
I I install basflight to the Naze, then, I get good signals and good debug info

Uh1Pilot, 2015-11-04:
It's still kind of confusing, you're using baseflight configurator to try
and setup cleanflight? Cleanflight does ports on the naze a little
different than baseflight. You have to tell it what the port is being used
for; then tell it what kind of gps you are using. You are probably going to
have to post a cli dump from cleanflight configurator (with cleanflight
firmware loaded) for anyone to help you.

jkmrs, 2016-01-18:
I have the same problem - though I only use Cleanflight. GPS gets a fast lock even in a concrete building when connected to the u-center via FTDI, but when running via Cleanflight and Flip32, I get the GPS detected by CF, some bars showing sats reception, but no fix, even outdoor.
I tried both via CF as well as u-center via Flip/gpspassthrough.
Any idea or suggestion?

mosi0815, 2016-01-30:
Same problem here. Maybe there's something wrong with the ublox auto configuration code. I'll now disable auto config and do it manual. I tried it with a NEO 7 and a NEO 8.

Arakon, 2016-02-20:
I see the same here. If I manually configure the GPS, things work sometimes, but my cheap GPS loses the config after an hour or so again.

iforce2d, 2016-03-06:
Same problem here. With 1.11 almost every row in the GPS signals display has a green bar in it, of strength 3 or greater. But number of sats shows zero and GPS fix remains as false forever. Interestingly the lat/lon display is shown correctly.

With 1.12 I can only see green bars showing occasionally, then they disappear. They never stay for more than a few seconds and never more than about 4-5 at a time if I'm lucky.

When I connect the GPS module to u-center it works very well every time.

Sometimes I find that after connecting the GPS module to u-center, if I connect it to the quad again I get a GPS lock and I can fly a few batteries with good GPS performance. But the next day, I have to do it all over again.

I would go back to 1.9 which worked beautifully, but this rev6 board needs 1.11 or newer :(

Arakon, 2016-03-06:
I'm wondering now if there's a code issue within the F1 code only.
On my SpracingF3, I get a working GPS fix fairly quickly.
I tested a CC3Df3 with GPS the other day (CC3D with a STM32F3 chip soldered to it and a special Betaflight version) and that also got a good fix after some minutes. On this exact same CC3D, before its CPU was replaced, I never got a valid fix (just the bars constantly appearing/disappearing).

Trussroads, 2016-03-06:
I am having a similar issue with an SPRacingF3 board running 1:12 and where I cant get GPS reception with FC but do on u-centre directly. I have not been able to get a Fix since building? I am on my second go around with receiver set-up now and after confirming correct settings on u-centre, then taking the quad into an open field on a cloudless day, I have been only able to recognise 3 sats at a time and with No bar movement and no Fix? although the Sat id number changes suggesting there is some level of visability? The reciever i'm using is a brand new small footprint version (RY725AI 10Hz) Although I cant say for sure what the cause of my own trouble is yet, the description from 'iforce2d' sounds very much like my own findings so I wanted to put my experience forward as well. 
Thanks for all the great dev work you all do on this software.. Cheerz!

Arakon, 2016-03-06:
Okay, I just did a direct comparison.
- 2x the same GPS unit (CJCMU Neo-6M)
- 2x a CC3D, one stock, one with F3 chip
- Both on the same Betaflight (2.4.1), since there's no CF version for the CC3DF3
- Both hooked up to PC using a CP2102 adapter on the main port.
- GPS on flex port, 38400 baud, UBLOX protocol, EGNOS
- gps_auto_config on, gps_auto_baud off
- All other settings stock

F3 unit got a 3D fix within 2 minutes with 4-6 sats (indoors on my desk).
F1 unit has been sitting there for 15+ minutes with no fix.

I then swapped the GPS units (and powered down the FCs for it of course). F1 unit with the GPS of the F3 unit got an immediate fix, then proceeded to lose the fix over the next 30 seconds (bars kept dropping to 0 more and more) and has not been able to get a fix again since.
Same GPS unit back to F3: Fix in seconds.

digitalentity, 2016-03-06:
This issue was encountered in iNav branch and it was concluded that F1 chip is giving too much electro-magnetical noise. Some GPS units are more immune to this, some are less.

Arakon, 2016-03-06:
In the surrounding area or what? Because I had the CC3Ds sitting ~10cm next to the GPS units and not below them (as would be usual in a copter) and both active at the same time and still got a fix on F3 but not F1. Or do you mean it's causing noise on the actual serial lines?

digitalentity, 2016-03-06:
@Arakon, in the surrounding area mostly. Shielding and ferrite ring on the wires going to GPS seem to help a little.

digitalentity, 2016-03-06:
F3 doesn't seem to affect GPS as much as F1.

iforce2d, 2016-03-07:
hmm... interesting. I can confirm the previous few posts about noise/interference, for F1 at least. I was skeptical until I tried it myself, but you can see what happens here:
https://youtu.be/H4Be_991D_s
I can also confirm that some GPS modules are less susceptible than others.
Can't say anything about the F1/F3 difference but I'll take your word for it :)

Trussroads, 2016-03-07:
Will review my own for signal isolation as well.. Thanks for following up on this. I hope to have good results with mine? Cheers

Regards,
Scott..

Arakon, 2016-03-07:
Curious. I just tried with a CC3D.. just powering it near the GPS doesn't seem to have any influence on the signal, but if the GPS is actually connected to the CC3D, my signal goes to hell.
In fact, even the connection to the GPS itself gets dropped at times. On F3, no such issue.

scu319hy, 2016-03-08:
disable NEMA messages except GxGGA/GxRMC/GxGSV.
disable GLONASS, BeiDou, Galileo...
GPS(NEMA mode) works well. it won't fix if other NEMA message exists.

Arakon, 2016-03-08:
As far as I know, the autoconfig disables them already. It also doesn't explain why the same firmware with the same GPS code works fine on one chip, and not the other, even using the same GPS unit.

KP1533TM2, 2016-06-18:
I'm having similar problems with my new uBlox M8N off eBay. It's connected to my Naze32 in my quadcopter, and using gpspassthrough I can then connect it to uCenter app and see that it gets the 3D fix with little trouble quite quickly (receiver is mounted 2cm above Naze32). However, if I powercycle everything, and go to my FC with Cleanflight configurator, I can see a lot of long green bars in GPS tab, but Qty values always stay at 0 and 3D fix never happens. I've tried setting up my GPS according to documentation, that didn't help.
Some radio interference from Naze32 does take place, but not to such extent I feared: reception on a quadcopter was definitely a bit worse than without it, but as I stated above, uCenter got 3D fix without much trouble in either case.

Are systems other than GPS treated differently by the receiver/FC? I specifically bought M8N for the fact that it can get better fix due to more satelites from different systems simultaneously (and for the fact this particular board has EEPROM to store settings), and according to scu319hy, I have to disable anything but GPS?
Bummer :(

P.S. Before that, I had an uBlox 7, and while it was flaky while being mounted on the quadcopter, it could actually acheive 3D fix with Naze32. But indeed, uBlox 7 is a single-system receiver, so it was set to receive GPS at the time...

jflyper, 2016-07-21:
Do people still have this problem? Take a look at #2319

KP1533TM2, 2016-07-21:
Compiled the latest master now, still have the same problem. No fix on M8N. Went through the GPS setup guide again, it didn't help. I noticed something odd, though: besides those UBX packets that I was supposed to turn on, I also see a lot of NMEA packets (that, according to the guide, should be off).
I'm trying to turn them off, and as I'm selecting the packet type from the drop-down box (on the MSG page), a moment later it automatically selects the previous packet type back. I figured that happened because of auto-polling, so I unchecked the corresponding button, and now I can select stuff, and I can "Send" it, but the GPS doesn't seem to notice - all NMEA packets that I tried to disable are still there (even after save/powercycle), and I have no clue what to do about it. I'm starting to think that I've messed up my config somehow, or maybe I'm missing something really simple. That GPS type is in the compatibility table on appropriate documentation page, but somehow I can't get it to work.

KP1533TM2, 2016-07-22:
Well, that's embarassing. I managed to get my GPS working today, with all the systems, it seems.
Turns out, 'gpspassthrough' is glitchy in a way that prevents GPS from being configured in u-Blox.
Today I connected my GPS to an Arduino as a UART-USB adapter (disabling the AVR CPU by shorting /RESET to GND, too lazy to buy a separate adapter), fired up my Windows VM with u-Blox, went through the GPS setup guide again, and voila! Getting 5..9 satellites on my window with a stable, almost solid, and maybe a bit drifty 3D fix.
This time I didn't even need to turn off systems other than GPS :)

So yeah, considering this problem solved for myself. The question "why it didn't work through FC" still may remain though.

closed 2017-02-25 by hydra