cleanflight/cleanflight #1413: Issue in Hott telemetry synchronization

Pierre-A, 2015-10-19:
In order to allow hardware serial support for Hott telemetry (issue #1021), a workaround has been introduced in the Hott process (commit 5142FF0 – august 19th) : in procedure *hottSendTelemetryData* in file *main/telemetry/hott.c*).

    ....
    if (!hottIsSending) {
        hottIsSending = true;
        // FIXME temorary workaround for HoTT not working on Hardware serial ports due to hardware/softserial serial port initialisation differences
        //serialSetMode(hottPort, MODE_TX);
        closeSerialPort(hottPort);
        hottPort = openSerialPort(portConfig->identifier, FUNCTION_TELEMETRY_HOTT, NULL, HOTT_BAUDRATE, MODE_TX, SERIAL_NOT_INVERTED);
        hottMsgCrc = 0;
        return;
    }

    if (hottMsgRemainingBytesToSendCount == 0) {
        hottMsg = NULL;
        hottIsSending = false;

        // FIXME temorary workaround for HoTT not working on Hardware serial ports due to hardware/softserial serial port initialisation differences
        //serialSetMode(hottPort, MODE_RX);
        closeSerialPort(hottPort);
        hottPort = openSerialPort(portConfig->identifier, FUNCTION_TELEMETRY_HOTT, NULL, HOTT_BAUDRATE, MODE_RX, SERIAL_NOT_INVERTED);
        flushHottRxBuffer();
        return;
    }
    ....

This broke part of the data transmission resulting in non desired behaviour (see last comments in issue #260).

The pictures below illustrate that (scope on RX telemetry pin).
I used Cleanflight commit 0150ab (october 15th) compiled with or without the workaround. 

**pict 1 :** previous code is uncommented and workaround is removed.
As required by the Hott standard, the Hott binary request is followed by a 3ms delay before the data are transmitted. This is repeated each 200ms and is the normal response for data transmission (values and alarm parameters are correctly decoded by the Hott TX).
[image]

**pict 2 and 3 :** workaround active, as already merged in Cleanflight.
One can see the effect of the duration of the workaround code : the data packet is delayed by 50ms after the request and overlap the next request. This induces a desynchronization and so decoding issues.
[image]
[image]

So it is clear this workaround is not a real solution and one has to work a bit more.

**@hydra** : what do you think ?
**@NOS-copter** : I suppose you now have telemetry through an uart port, don’t you observe any decoding issue as mentionned by johnlenfr in #260 ?

NOS-Copter, 2015-10-19:
I do not have any copter using the workaround. i just found that i get data with the workaround and wanted to share my findings. i dont have a scope and never dug in the real issue.
i think the workaround (close and reopen the Serial) did just work on a hardwareSerial. So it was never intended for the use of softserial.

@Pierre-A is the RX telemetry pin on a software or hardwareSerial?

ledvinap, 2015-10-19:
@pierre-A: 50ms delay is huge. closing and opening serial port is suboptimal, but it should take at most few us... Something other is happening..

Pierre-A, 2015-10-19:
**@NOS-copter** : softserial
**@ledvinap**, yes for sure, this is the effect of. One still has to explain it.

ledvinap, 2015-10-19:
Hmm ... softserial is heavily broken ... the delay may be related to wrongly initialized time compare register - usually it causes 65.535ms delay, but it may be shorted (or referenced to some old edge)

Pierre-A, 2015-10-20:
It seems that the Hott telemetry is mainly used on a softserial port due to its low rate.
So, while this timing issue is not solved and as a short-term measure, I would suggest to comment this workaround and come back to the previous code. 
Any comment, idea ..?

ledvinap, 2015-10-20:
Just apply close-open only if port is UART ...

Pierre-A, 2015-10-20:
Yes, already did it and things work fine, but it is just a workaround over another workaround ...
I'am afraid there is no other solution right now and Hott telemetry on uart is to be tested (**@NOS-copter**, did you do it ?).

NOS-Copter, 2015-10-26:
no did not test... my idea was to rewrite the softserial for a single-wire support. one could save one pin, no soldering and everything is easier. but atm i'm not good enough/experienced with CF to do such task. i will see what my next experiences will bring.
@Pierre-A i just flashed 1.10.0 and reverted to the setMode in hott.c but get no response on softSerial (hardware is same which worked fine on CF1.9). are there more changes needed in 1.10?

Pierre-A, 2015-10-26:
**@NOS-Copter**, did you look at last commit in PR #1419 ? hott works fine for me on sofserial1

NOS-Copter, 2015-10-26:
yes i saw it and changed that lines, but it doesnt work.
if i copy the whole hott.c i get an error because of serialRXBytesWaiting is not existing?!

ledvinap, 2015-10-26:
@NOS-Copter : Good luck fixing softserial ;-)
I have softserial rewrite in my local branch (https://github.com/ledvinap/cleanflight/blob/local/src/main/drivers/serial_softserial.c).
It is capable of  SBUS (100kbps) + Smartport (57kbps), with few errors (I can echo back 1MB of data at 115200 with less than one error on average).
But quite radical redesign of cleanflight was necessary ...

ledvinap, 2015-10-26:
@NOS-Copter 
serialRXBytesWaiting was probably renamed by recent commit, maybe you are using older version?

NOS-Copter, 2015-10-26:
just downloaded the 1.10.0 release and then wanted to get hott working ;)

ledvinap, 2015-10-26:
You should probably switch to master.. 
serialTotalBytesWaiting was (probably) the old name ..

NOS-Copter, 2015-10-26:
now i took the master and took the hott.c of PR #1419 
same result. no data :(

Pierre-A, 2015-10-26:
don't understand that, as I used the master *efc31f9* where serialRXBytesWaiting is defined in file serial.c.
And this is compiled as usual.
(serialTotalBytesWaiting is an old name)

NOS-Copter, 2015-11-03:
So I'm sitting in front of an scope and I can see no data coming if using Softserial. Everything is fine, if i use Hardware serial: data is received and answer gets transmitted.
I have HottDebug enabled, so i can ensure that softserial gets the requests and goes into the right SetMode function. only it isnt transmitting anything.

@Pierre-A perhaps u can make a zip with source or at least a hex i can try. I did even try the master u used and there is nothing showing.

PS: if i change to FrSky telemtry i get data from softserial, so the write data seems to work. Really dont know how to proceed...

Pierre-A, 2015-11-03:
Which cleanflight version do you use ?
As PR #1419 has been merged in Cleanflight master, you can compile last commit (f18936a).
I use this one and Hott telemetry works fine on Softserial. Are you sure the diode is right way (cathode to softserial Tx) ?

Pierre-A, 2015-11-08:
This should be closed

closed 2015-12-21 by Pierre-A